nickbabcock/OhmGraphite #1: Monitoring data is not available from some machines

arielcross, 2018-03-05:
I'm having an issue with one of the machine I have ohmgraphite installed and running as a service but at my graphite server there's no data from the certain machine.

No firewall (also on linux server the firewall and SELinux has been disabled).
No antivirus.
The IP address pointer is correct.

I would like to know if there is anything I missed to configure.

nickbabcock, 2018-03-05:
There should be a `OhmGraphite.log` file in the same directory of where `OhmGraphite.exe` is.

Almost the entire log should be filled with a line like:

```
2018-03-05 00:48:57.8576|INFO|OhmGraphite.MetricTimer|Sent 39 metrics in 13.5089ms
```

Notice that the line contains "INFO". Are there any lines in your log that contains "ERROR"?

arielcross, 2018-03-06:
[image]

Here's the log it sent data to the right address but the server didn't received anything.
There should be a lot of data already but this looks very weird.....

nickbabcock, 2018-03-06:
Thanks for the info. Next thing to try.

- Open up powershell in the directory that you downloaded OhmGraphite to
- `.\OhmGraphite.exe stop` -- to make sure that the service isn't running
- `.\OhmGraphite.exe run` -- to run the command interactively.

What output do you see?

OhmGraphite is working correctly on your other machines, but not this one? 😅

arielcross, 2018-03-07:
[image]

Here's the output but still no luck :(

Other machines work fine except this one.....

nickbabcock, 2018-03-07:
I released [0.1.3](https://github.com/nickbabcock/OhmGraphite/releases/tag/v0.1.3), which will give us more information. 

- From 0.1.3 copy over `OhmGraphite.exe` into your installation
- Edit `NLog.config`

Replace

```xml
  <rules>
    <logger name="*" minlevel="Info" writeTo="file" />
    <logger name="*" minlevel="Info" writeTo="console" />
  </rules>
```

with

```xml
  <rules>
    <logger name="*" minlevel="Debug" writeTo="file" />
    <logger name="*" minlevel="Debug" writeTo="console" />
  </rules>
```

The hope is that the log displays more information and we can pinpoint the issue.

arielcross, 2018-03-09:
[image]
Here's the latest feed back.
Ohmgraphite is working correctly but for some reason the data didn't reach the server :(

Is there a way I can remove ohmgraphite from the registry ?

nickbabcock, 2018-03-09:
Did you make sure to set `minlevel="Debug"` in `NLog.config`, as there should be a lot more output.

It's ok, we gave it a good effort. One day the mystery might be solved -- sorry for the hassle 😄 

To uninstall

```
.\OhmGraphite.exe uninstall
```

Then you can delete the installation directory.

arielcross, 2018-03-13:
[image]
Here's with the debug on.

nickbabcock, 2018-03-13:
Looks like `.\OhmGraphite.exe restart` is being executed, which is not a valid command. To restart OhmGraphite, execute 

```
.\OhmGraphite.exe stop
.\OhmGraphite.exe start
```

nickbabcock, 2018-04-04:
If ever you want to give this another shot, let me know. In the meantime, I've documented how to uninstall in the readme.

closed 2018-04-04 by nickbabcock