nickbabcock/OhmGraphite #7: Crash in x86 Win7sp1Ultimate When Strart the OhmGraphite.exe

Tea-NT, 2018-07-19:
Here is exception info：

> Topshelf.HostFactory Error: 0 : An exception occurred creating the host, System.TypeInitializationException: The type initializer for 'OhmGraphite.Program' threw an exception. ---> System.MissingMethodException: Method not found: 'Int32 System.Environment.get_CurrentManagedThreadId()'.
   at NLog.LogFactory.<GetDefaultCandidateConfigFilePaths>d__84..ctor(Int32 <>1__state)
   at NLog.LogFactory.GetCandidateConfigFilePaths()
   at NLog.LogFactory.TryLoadFromFilePaths()
   at NLog.LogFactory.get_Configuration()
   at NLog.LogFactory.GetLogger(LoggerCacheKey cacheKey)
   at NLog.LogFactory.GetLogger(String name)
   at NLog.LogManager.GetCurrentClassLogger()
   at OhmGraphite.Program..cctor()
   --- End of inner exception stack trace ---
   at OhmGraphite.Program.<>c.\<Main\>b__1_1(ServiceConfigurator`1 s)
   at Topshelf.ServiceExtensions.CreateServiceBuilderFactory[TService](Action`1 callback)
   at Topshelf.ServiceExtensions.Service[TService](HostConfigurator configurator, Action`1 callback)
   at OhmGraphite.Program.<>c.<Main>b__1_0(HostConfigurator x)
   at Topshelf.HostFactory.New(Action`1 configureCallback)
Topshelf.HostFactory Error: 0 : The service terminated abnormally, System.TypeInitializationException: The type initializer for 'OhmGraphite.Program' threw an exception. ---> System.MissingMethodException: Method not found: 'Int32 System.Environment.get_CurrentManagedThreadId()'.
   at NLog.LogFactory.<GetDefaultCandidateConfigFilePaths>d__84..ctor(Int32 <>1__state)
   at NLog.LogFactory.GetCandidateConfigFilePaths()
   at NLog.LogFactory.TryLoadFromFilePaths()
   at NLog.LogFactory.get_Configuration()
   at NLog.LogFactory.GetLogger(LoggerCacheKey cacheKey)
   at NLog.LogFactory.GetLogger(String name)
   at NLog.LogManager.GetCurrentClassLogger()
   at OhmGraphite.Program..cctor()
   --- End of inner exception stack trace ---
   at OhmGraphite.Program.<>c.\<Main\>b__1_1(ServiceConfigurator`1 s)
   at Topshelf.ServiceExtensions.CreateServiceBuilderFactory[TService](Action`1 callback)
   at Topshelf.ServiceExtensions.Service[TService](HostConfigurator configurator, Action`1 callback)
   at OhmGraphite.Program.<>c.<Main>b__1_0(HostConfigurator x)
   at Topshelf.HostFactory.New(Action`1 configureCallback)
   at Topshelf.HostFactory.Run(Action`1 configureCallback)   

I tryed  ``` OhmGraphite.exe start```  ,```OhmGraphite.exe run``` and  ```OhmGraphite.exe install```,all of them was not work.
The  OhmGraphite.exe work well in x64 Win7

Tea-NT, 2018-07-19:
I installed .NET Framework 4.7.2 to solved the question,
I think add .NET Framework request in README.md file is batter.
and thanks for your opensource code.

I google this page:
[System.MissingMethodException Int32 System.Environment. get_CurrentManagedThreadId()](https://stackoverflow.com/questions/15569664/system-missingmethodexception-int32-system-environment-get-currentmanagedthread)
so i download the newest .NET Framework to try.it works.

nickbabcock, 2018-07-19:
You're right, .NET 4.6 is required to run. Hopefully most people won't require downloading a newer version as Windows 10 includes it by default.

I'll close this when I've added it to the readme, thank you very much for your report.

Tea-NT, 2018-07-20:
Yeah，you’re right，so i spend some time to solve it.

I deployed the OhmGraphite in my Intel Celeron J1800 CPU Clients,OhmGraphite can not collect the CPU temperature,only SSD temperature,if running Open Hardware Monitor it was fine. 

[image]
I want to try  remote debugging to find out the reason,but there is not Remote debug checkbox in OhmGraphite project's Properties --Debug page with VS2017 .I am confused.

[image]

nickbabcock, 2018-07-20:
This should be fixed now on the master branch. Are you able to try it out? You seem savvy! If not, I can cut a pre-release version.

I did notice that I occasionally would have 70 or 71 metrics reported, and I didn't think much of it. I took a deeper look into how open hardware monitor uses its library and I found some inconsistencies between how I use it and the gui. I wouldn't think these inconsistencies would cause only some metrics to be detected on startup.

Anyways, the latest code successfully detects when new sensors are available. I saw that 5 minutes after starting I went from 70 metrics to 71 (why there is a such a delay in detection, I don't know)

Tea-NT, 2018-07-21:
Thanks for your speedy response!
Yesterday after I configed Grafana and then try see the code，our office and near street region was out of power because of a transformer failure(weather is too hot,in outside temperature may over 40℃ about 104℉ ),electricity power need a few days to restore.

I will test fast as i can.

nickbabcock, 2018-07-21:
No worries, no rush 😄

Tea-NT, 2018-07-24:
Hey!
I think i find the reason,today i find the way to remote debug the OhmGraphite by use the ```Attach to Process```  in VS2017.
First,copy the Remote Debuger in VS2017 \Common7\IDE\ from install folder to Client PC,then start the ```msvsmon.exe```.
After that,in debuger PC use the  ```Attach to Process```,input the remote client ip and select OhmGraphite  Process.
[image]

All sensors was added through ```HardwareAdded``` event function,but next step in ```ReportedValues``` function two CPU temperature sensor and a other sensor has no values.
[image]
[image]

that is why my client PC only sent 8 metrics.

I am a novice in .NET,and not familiar with Nlog,if the log output ERROR message that don't need remot debug to find out that.

nickbabcock, 2018-07-24:
> Left pic is only start OhmGraphite,right is start OhmGraphite after ohm startup.

Thank you for the info and all the debugging. That is certainly weird that OhmGraphite would see more metrics when ohm is running 😖 

And just to triple check, are you running OhmGraphite and ohm as administrator?

Tea-NT, 2018-07-24:
Yes,ohm need administrator permission when startup,each time i also tested Right Click -- Run As Administrator option to run OhmGraphite.exe. Still the same result.

It look ohm activated the others sensors.After OhmGraphite  startup whth ohm. Even close the ohm, OhmGraphite also work fine,it sent 15 metrics. If then restart OhmGraphite,change back to sent 8 metrics.

Tea-NT, 2018-07-27:
Ok,I give up,Thanks your response anyway.

I tried add a ```app.manifest``` file in OhmGraphite Project to request ```Administrator``` permission,it failed.
the last way only start OhmGraphite in Windows Services set ```Automatict(Delayed Start)```,wait ohm startup.

nickbabcock, 2018-07-27:
Ah shoot, yeah it's a bummer.

I'll close this issue, but if anyone wants to continue working on this issue (or has this issue), feel free to respond (or open a new issue).

closed 2018-07-27 by nickbabcock