nickbabcock/OhmGraphite #25: Dashboards|Units

wuast94, 2018-12-19:
Are there any Dashboards available for Import in Grafana? would make a lot things easier :) or a doku that shows what measurement is what unit ? i get raw numbers but cant interpret them without knowing what units are used. And thanks for your work :)

nickbabcock, 2018-12-19:
You're right! I should feature importable dashboards that can be made with OhmGraphite. The one downside is that each datasource would require its own dashboard and dashboards are subjective to tastes. But something is better than nothing.

As far as units: those should be documented, but in the meantime the [source code should be enlightening](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/4ae6141b31ba875ed78e9477d699c5ac31748c2e/Hardware/ISensor.cs#L18-L31)

wuast94, 2018-12-19:
One dashboard with all graphs should be right. Anyone can copy and edit things as needed. 

Different Data sources should be doable with variables I think :)

wuast94, 2018-12-20:
Never did this before but i think i have made a pull request with a json file for import in Grafana what should work :)

Nightreaver, 2020-01-16:
I wonder... beside having checked the source... how to make sense of Throughput data for network devices.

Taskmanger is showing me... 6-9mbit Down, and 0 - 2mbit up speed, changing a lot of course.

But the data in graphite shows download as around ~100 "k" (Unit "short") if I select "mean" as field value.
I played around but cannot find any combination that is getting close to the 10mbit. 
If i change the unit to bytes/sec, I end up with the similar ~100 "kb" per second which still feels wrong.

Any Idea? Recommendation? 
wuast94 dashboard sadly doesnt contain throughput ...

wuast94, 2020-01-16:
You can add math and devide or multiplied. Set the unit you want to (for example MB/s) and do math until it gets how you want it

wuast94, 2020-01-16:
I don't know in which unit it's get extracted to influx db so I dont know which factor you should use. And you should calculate correct (1MB 1024 Byte) etc.. That can make a big difference too

nickbabcock, 2020-01-17:
Throughput is bytes per second.

I did a quick test where I downloaded a steam game and confirmed that the throughput in the dashboard (measured in MB/s) matched the steam ui, so I don't think there is a calculation issue.

My guess is that you may have more than one network device, so taking the average will make it lower. For instance, I have ethernet, wifi, and bluetooth. Make sure you have one device selected to see the correct throughput. If this isn't the case, let me know and we can see troubleshoot some more.

I do agree that it would be immensely helpful if I provided the dashboards I use. They may not be perfect for everyone, but at least they would provide starting point and I can keep them updated

wuast94, 2020-01-17:
I cant realy test it becouse i dont have any device with multiple networkk interfaces. 

But im running telegraf on my server and have there a throughput graph. and this project use telegraf too for that as far as i know. 

i attach a screenshot from this dashboard so u can check it if this work for you. and on my server i have a lot of interfaces (every docker container creates on and some more from other things) but i seleced the one i want, dont know if u do that. 

[image]

and here the json: 
[https://pastebin.com/EL1i1SNg](url)

nickbabcock, 2020-01-22:
I'm going to close this issue. I've uploaded dashboards (prometheus, influx, graphite, postgres -- listed in the readme) to grafana that demonstrate some of the graphs one can make. Further suggestions are welcomed.

closed 2020-01-22 by nickbabcock