nickbabcock/OhmGraphite #43: Prometheus requires Metric Name sanitizing

Tropfchen, 2019-02-23:
Some of names that Open Hardware Monitor spits out (in my case `nic_{hexhex-hex-hex-hex-hexhex}_data_2`) don’t pass `^[a-zA-Z_:][a-zA-Z0-9_:]*$` regex check performed by `Prometheus.MetricFactory.CreateGauge` for name, causing exception and http error 500.

Adding some additional character removing/replacing at the end of https://github.com/nickbabcock/OhmGraphite/blob/7f752eea7ea12c69b0d237c766756b025e569813/OhmGraphite/PrometheusCollection.cs#L33 does solve the problem. 

And by the way, thank you very much for writing this tool!

nickbabcock, 2019-02-23:
Very good spot! Yes, I hadn't tested prometheus with the new NIC sensors. Is this something you'd want to fix and test out (and issue a PR for)?

Tropfchen, 2019-02-24:
I would prefer not to - all my experience with C# was running unit tests and testing out if replacing {}- chars is enough to run this code without problems on my machine haha

nickbabcock, 2019-02-25:
No worries, I'll have the fix in a day or two, I just didn't want to preempt a potential contribution 😄

nickbabcock, 2019-02-25:
v0.8.2 has been released, give it a try!

closed 2019-02-25 by nickbabcock

Tropfchen, 2019-02-26:
Works great, thanks! I just hope there won’t be users with sensor identifier that still won’t pass the test

PR #44 (2019-02-25): Fix prometheus and nic guids
`{`, `}`, and `-` are all invalid characters for a prometheus metric so we remove them or replace them with a `_`. Closes #43