nickbabcock/OhmGraphite #53: Hostname capital and lowercase collection problem

JinkunTian, 2019-04-06:
My computer‘s hostname is 'Inspiron-3568' ,'I' is in capital, ohmGraphite collect “INSPIRON-3568” as the hostname and exports “INSPIRON-3568” to the InfluxDB.I used ohmGraphite  and Telegraf at the same time to collect data.This problem caused a host to have two pieces of information when I used hostname to distinguish devices,How can i deall this ? And thanks for your OhmGraphite :)

nickbabcock, 2019-04-06:
Yeah I've experienced similar issues. It boils down to the difference between a NetBIOS name and DNS name.

```c#
using System.Net;

Environment.MachineName // NetBIOS
Dns.GetHostName()
```

prints

```
TINI
Tini
```

I think I'll need to add a configuration option for one to opt into using the DNS name instead.

JinkunTian, 2019-04-06:
You are right! Add a configuration option for one to opt into using the DNS name or  NetBIOS name is best way to solve this .  Thank you for your help : )

nickbabcock, 2019-04-09:
This has been fixed in 0.8.4 with the `name_lookup` option set to `dns`

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="name_lookup" value="dns" />
  </appSettings>
</configuration>
```

closed 2019-04-09 by nickbabcock