nickbabcock/OhmGraphite #68: Combine Ati and Nvidia GPUs

nickbabcock, 2019-05-23:
For hardware types there are multiple GPU brands listed (ATI and Nvidia). This distinction is only for GPUs (ie: there is only one cpu hardware type -- it's not broken into amd and intel). The consequence of this distinction is that if one were to swap out their GPU for another brand, then all metric names would need to be migrated over or one would need the foresight to cover both use cases like:

```sql
SELECT * FROM ohm_stats
    WHERE hardware_type = 'GpuNvidia' OR
                 hardware_type = 'GpuAti'
```

A potential solutions would be to simply export both of these metrics under the single label of "gpu". This would be a breaking change to all users, something I try very hard to avoid.

So I'll leave this issue open and if there has been significant feedback or if it is bothersome, then I can look implement something (ie: a config flag).

swiffer, 2023-01-09:
i just noticed that some metrics have been empty in my grafana dashboard after switching my gpu (nvidia -> amd).

turns out there are currently some queries missing it's amd counterparts in the grafana dashboard query config.

i was able to add amd queries for memory and fan, data rate seems not to be available

[image]

having only one label with `gpu` would make running multiple queries (one per vendor) obsolete, right?

nickbabcock, 2023-01-09:
Ah, you're right, it's an oversight that several of the gpu graphs leave off their amd counterpart.

Consolidating them could fix the issue, though data rate would still be blank. I think I've been hesitant to combine them for systems that run both amd and nvidia gpus simultaneously, but I think this situation would be exceedingly rare -- though probably possible to workaround with deterministic ordering (like nvidia gpus before amd).

swiffer, 2023-01-09:
What Happens with Intel Arc gpus / when there is more than one Nvidia GPU installed?

nickbabcock, 2023-01-09:
Intel arc gpus are currently unsupported https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/792

But that is a great point, instead of exposing a third type of GPU when support is added, it would be nice if the dashboards were already generic over a consolidated GPU type, so nothing would need to change.