nickbabcock/OhmGraphite #70: Using Prometheus

Stopi, 2019-05-30:
Hi,

I couldn't find a way to get metrics from Prometheus.
It's not really a bug, but any help would be greatly appreciated, as I didn't know where to ask for support.

OhmGraphite is installed on Windows Server 2012 R2, with this config:
```xml
<?xml version="1.0" encoding="utf-8"?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="4445" />
    <add key="prometheus_host" value="127.0.0.1" />
    <add key="tags" value="true" />
  </appSettings>
  <startup>
    <supportedRuntime version="v4.0" sku=".NETFramework,Version=v4.6.1" />
  </startup>
</configuration>
```
Service is running and according to netstat, the process is listening on port 4445.
Firewall has open this port for TCP in & out.

Then, on the Prometheus machine, I've configured:
```yml
scrape_configs:
  - job_name: 'node'
    static_configs:
    - targets: ['winsrv:4445']
      labels:
        host: 'coin'
```
So far I haven't seen any error in the log.
But when asking Prometheus for metrics on `{host="coin"}`, I get only:
```
scrape_duration_seconds{host="coin",instance="winsrv:4445",job="node"} | 8.581074809
scrape_samples_post_metric_relabeling{host="coin",instance="winsrv:4445",job="node"} | 0
scrape_samples_scraped{host="coin",instance="winsrv:4445",job="node"} | 0
up{host="coin",instance="winsrv:4445",job="node"} | 0
```
However, I'm sucessfully monitoring with `wmi_exporter` on the same machine.
What am I missing here ?

nickbabcock, 2019-05-30:
```xml
 <add key="prometheus_host" value="127.0.0.1" />
```

If your prometheus instance is running on a difference server, you'll want to change this from 127.0.0.1 to something like `<internal-ip-here>` or `*`

Stopi, 2019-05-31:
Hi, thanks for your answer but it doesn't help.
At first I was using
```xml
<add key="prometheus_host" value="*" />
```
and already have the same issue.

I also did try use the prometheus server IP.
But then it's worse: the ohmgraphite service doesn't start at all.

nickbabcock, 2019-05-31:
Maybe logs will help. 

- Go to where you installed ohmgraphite, delete `OhmGraphite.log`
- Restart OhmGraphite and ensure your config has `<add key="prometheus_host" value="*" />`
- See if there are any errors / post the log here.

Stopi, 2019-06-05:
No luck, as I mentioned there's nothing to see in the logs.
But I've followed your steps anyway and here is the complete log:
```
2019-06-05 10:25:24.7468|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Stopping
2019-06-05 10:25:24.8718|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Stopped
2019-06-05 10:25:27.2690|INFO|OhmGraphite.Program|Prometheus port: 4445
2019-06-05 10:25:27.5314|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2019-06-05 10:25:27.5470|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.0.194, .NET Framework v4.0.30319.42000
2019-06-05 10:25:27.5783|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|Starting as a Windows service
2019-06-05 10:25:27.5783|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Starting
2019-06-05 10:25:28.7530|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Started
```

nickbabcock, 2019-06-05:
Ok so far so good. In `NLog.config` change the following line

```xml
    <logger name="*" minlevel="Info" writeTo="file" />
```

to

```xml
    <logger name="*" minlevel="Debug" writeTo="file" />
```

There should be a lot more output. When prometheus goes to scrape metrics the following text will appear in the log:

```
prometheus update metrics
```

If said text is not in the file that means that prometheus is unable to access the server. I have a config of 

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="4445" />
    <add key="prometheus_host" value="*" />
  </appSettings>
</configuration>
```

and when I go to http://localhost:4445/metrics in a browser on the local machine, I can see all the metrics.

Is this the same for  you?

Stopi, 2019-06-06:
OK, so I can see all of it.
All metrics are visible at http://localhost:4445/metrics and the debug outpu in log shows:
```
2019-06-06 07:45:18.5113|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2019-06-06 07:45:18.5663|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2019-06-06 07:45:21.2515|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2019-06-06 07:45:21.2515|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2019-06-06 07:45:33.5052|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2019-06-06 07:45:33.5052|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2019-06-06 07:45:36.2523|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2019-06-06 07:45:36.2523|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2019-06-06 07:45:48.5060|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
...
```
Looks like it works.
So I've tried again to ask Prometheus for `{host="coin"}`:
```
dotnet_collection_count_total{generation="0",host="coin",instance="<ip-winsrv>:4445",job="node"}	11
dotnet_collection_count_total{generation="1",host="coin",instance="<ip-winsrv>:4445",job="node"}	3
dotnet_collection_count_total{generation="2",host="coin",instance="<ip-winsrv>:4445",job="node"}	2
dotnet_total_memory_bytes{host="coin",instance="<ip-winsrv>:4445",job="node"}	2669216
hdd_0_load_0{app="ohm",exported_host="<host-winsrv>",hardware="ST3500418AS",hardware_type="HDD",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Used Space",sensor_index="0"}	35.5518112182617
hdd_0_temperature_0{app="ohm",exported_host="<host-winsrv>",hardware="ST3500418AS",hardware_type="HDD",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Temperature",sensor_index="0"}	28
intelcpu_0_clock_0{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Bus Speed",sensor_index="0"}	265.999450683594
intelcpu_0_clock_1{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #1",sensor_index="1"}	1595.99658203125
intelcpu_0_clock_2{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #2",sensor_index="2"}	1595.99658203125
intelcpu_0_load_0{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Total",sensor_index="0"}	7.67045021057129
intelcpu_0_load_1{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #1",sensor_index="1"}	6.25
intelcpu_0_load_2{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #2",sensor_index="2"}	9.09090614318848
intelcpu_0_temperature_0{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #1",sensor_index="0"}	39
intelcpu_0_temperature_1{app="ohm",exported_host="<host-winsrv>",hardware="Intel Core 2 Duo E7500",hardware_type="CPU",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="CPU Core #2",sensor_index="1"}	29
nic_4F27F629_E762_42CA_BA05_3BF9AA0BAC83_data_2{app="ohm",exported_host="<host-winsrv>",hardware="Ethernet",hardware_type="NIC",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Data Uploaded",sensor_index="2"}	1.91623282432556
nic_4F27F629_E762_42CA_BA05_3BF9AA0BAC83_data_3{app="ohm",exported_host="<host-winsrv>",hardware="Ethernet",hardware_type="NIC",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Data Downloaded",sensor_index="3"}	1.71517431735992
nic_4F27F629_E762_42CA_BA05_3BF9AA0BAC83_load_1{app="ohm",exported_host="<host-winsrv>",hardware="Ethernet",hardware_type="NIC",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Network Utilization",sensor_index="1"}	0.0416046157479286
nic_4F27F629_E762_42CA_BA05_3BF9AA0BAC83_throughput_7{app="ohm",exported_host="<host-winsrv>",hardware="Ethernet",hardware_type="NIC",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Upload Speed",sensor_index="7"}	30226.96875
nic_4F27F629_E762_42CA_BA05_3BF9AA0BAC83_throughput_8{app="ohm",exported_host="<host-winsrv>",hardware="Ethernet",hardware_type="NIC",host="coin",instance="<ip-winsrv>:4445",job="node",sensor="Download Speed",sensor_index="8"}	21778.80078125
process_cpu_seconds_total{host="coin",instance="<ip-winsrv>:4445",job="node"}	1.9375
process_num_threads{host="coin",instance="<ip-winsrv>:4445",job="node"}	9
process_open_handles{host="coin",instance="<ip-winsrv>:4445",job="node"}	366
process_private_memory_bytes{host="coin",instance="<ip-winsrv>:4445",job="node"}	35287040
process_start_time_seconds{host="coin",instance="<ip-winsrv>:4445",job="node"}	1559782357.67967
process_virtual_memory_bytes{host="coin",instance="<ip-winsrv>:4445",job="node"}	622571520
process_working_set_bytes{host="coin",instance="<ip-winsrv>:4445",job="node"}	43773952
[... cut ...]
scrape_duration_seconds{host="coin",instance="<ip-winsrv>:4445",job="node"}	0.016363228
scrape_samples_post_metric_relabeling{host="coin",instance="<ip-winsrv>:4445",job="node"}	29
scrape_samples_scraped{host="coin",instance="<ip-winsrv>:4445",job="node"}	29
up{host="coin",instance="<ip-winsrv>:4445",job="node"}
```
Now I'm wondering why it didn't work in the first place, since I changed nothing.
Maybe I had to wait more...

Everything looks good now, still not sure why.
Anyway thank you for your help.

Stopi, 2019-06-06:
OK, now I know what happend.
I made tests on several hosts and one of them hasn't got the rule in firewall to accept request on port 4445.
So it's entirely my bad!
Thank you anyway, now I know more ways to test this, the debug trick helped a lot.

closed 2019-06-06 by Stopi

nickbabcock, 2019-06-06:
Excellent, great to hear. Yes I must type up a debugging guide! 😄