nickbabcock/OhmGraphite #72: Cannot change password

Stopi, 2019-05-31:
I've configured 2 hosts to access grafana:
- one is for administrators (http://grafana)
- one is for public monitoring (http://monitoring)

Here is the Apache configuration:
```apache
<VirtualHost *:80>
    ServerName grafana

    ErrorLog /var/log/apache2/grafana-error
    CustomLog /var/log/apache2/grafana-access combined

    TimeOut 600
    ProxyPass / http://<grafana-ip>:3000/
    ProxyPassReverse / http://<grafana-ip>:3000/
    ProxyPreserveHost on
    RemoteIPHeader X-Forwarded-For
    RemoteIPTrustedProxy 127.0.0.1 ::1
    RequestHeader set "X-Forwarded-Proto" expr=%{REQUEST_SCHEME}
    RequestHeader set "X-WEBAUTH-USER" ""

    <ifModule mod_headers.c>
        Header unset ETag
        Header set Cache-Control "max-age=0, no-cache, no-store, must-revalidate"
        Header set Pragma "no-cache"
        Header set Expires "Wed, 11 Jan 1984 05:00:00 GMT"
    </ifModule>
</VirtualHost>
<VirtualHost *:80>
    ServerName monitoring

    ErrorLog /var/log/apache2/grafana-error
    CustomLog /var/log/apache2/grafana-access combined

    TimeOut 600
    ProxyPass / http://<grafana-ip>:3000/
    ProxyPassReverse / http://<grafana-ip>:3000/
    ProxyPreserveHost on
    RemoteIPHeader X-Forwarded-For
    RemoteIPTrustedProxy 127.0.0.1 ::1
    RequestHeader set "X-Forwarded-Proto" expr=%{REQUEST_SCHEME}
    RequestHeader set "X-WEBAUTH-USER" "public"

    <ifModule mod_headers.c>
        Header unset ETag
        Header set Cache-Control "max-age=0, no-cache, no-store, must-revalidate"
        Header set Pragma "no-cache"
        Header set Expires "Wed, 11 Jan 1984 05:00:00 GMT"
    </ifModule>
</VirtualHost>
```
With this configuration, I have an access for users without login on **monitoring** and a required login box on **grafana**, which is the desired behavior.

However, when connected as an admin, I cannot change my password in the **Preferences** interface, it says:
_You cannot change password when ldap or auth proxy authentication is enabled._ 

Locking the password change is expected for the public access, but not for the admin one.
I've tried to remove the line in apache vhost
```apache
# RequestHeader set "X-WEBAUTH-USER" ""
```
but this didn't resolve the issue.

I didn't configure any LDAP nor Auth Proxy, so the message confuses me.
Any advice on this ?

nickbabcock, 2019-05-31:
No, sorry I'm unable to help as I'm unfamiliar with that setup. I run grafana in docker behind nginx, and the admin password I have supplied via the `GF_SECURITY_ADMIN_PASSWORD` environment variable.

nickbabcock, 2019-06-04:
Closing this, let me know if you need further assistance

closed 2019-06-04 by nickbabcock

Stopi, 2019-06-05:
Sorry for delay to answer, I've been busy these days.
This issue is not related to Apache.
In fact Nginx could replicate the bug, as it's just a web server used as front proxy.
And this has nothing to do with the admin password or `GF_SECURITY_ADMIN_PASSWORD`.
Let me rephrase:
- I'm using automatic login on one host (**monitoring**) by sending the header `X-WEBAUTH-USER` and it works just fine. As expected, the user is automatically connected and cannot change his password because that doesn't make sens in a _non-login_ context.
- Now, on a second host (**grafana**) I don't use the `X-WEBAUTH-USER` header and so the login dialog prompt when you arrive on the site. This is fine too, but once connected, with _any_ user, I expect to be able to change the password. Which isn't possible because there's no form, just a message warning me: _You cannot change password when ldap or auth proxy authentication is enabled._

I hope this clarify the issue.

Stopi, 2019-06-05:
Oh dear! I'm so confused!
I'm posting on the wrong thread, this is a bug related to grafana and I just realize this is ohmgraphite's space.
Sorry for the noise, I'll post this on the correct place.