nickbabcock/OhmGraphite #114: Linux Alternative

palitu, 2020-01-13:
just to start, i love OhmGraphite.  Simple and quick method to track our infra, with a standardised set of metadata and structure!

I am looking for something that does the same thing, but for linux, preferably with the same naming conventions as OG, and ease of install/config

any ideas?

thanks

nickbabcock, 2020-01-14:
Thanks for stopping by! What is your use case (server, workstation, gaming) on linux?

While OhmGraphite is technically cross platform (it should be able to run Mono on linux), it's not what I use for my linux servers. I believe last time I checked there were significantly fewer metrics exported. So I sought different options.

On linux, I use a mixture of collectd (for ipmi, network connectivity, and S.M.A.R.T stats) and node_exporter for other system metrics for my prometheus server. It gets me most of the same metrics as OhmGraphite (my server lacks a GPU right now, so I can't give advice for that), but the metrics are named/structured differently, unfortunately, so I have separate dashboards for servers (linux) vs desktops (windows).

My long term plan (we're talking 2021, 2022) is to move to a more cross platform hardware sensor library so that OhmGraphite can provide consistent hardware metrics across a multitude of platforms.

palitu, 2020-01-20:
THanks Nick

It is a range of linux servers.  In my last job, i deployed a lot of ohmgraphite instances, and now we are utilising linux servers a lot more.

I just love the way that it is so easy to set up, and that all the data comes across in the same formating for dashboards.

I am thinking of trying to add an MQTT output to OG, so that we have a transport layer that can move across domains.  Will see if one of the devs can do a PR to add the capability.

(it will probably be something like a JSON or influx line protocol payload, to a custom topic).

cheers!

nickbabcock, 2020-01-20:
Interesting, I hadn't thought about a middleman between OhmGraphite and a given backend. If there are metric aggregators or proxies that use a format OhmGraphite already pushes (like the influxdb line protocol) that may be my preference. Else, ideally we'd leverage some sort of prior art to benefit the most people and minimize work and maintenance on our end (though I think there is something to be said for supporting output via mqtt, csv, amqp, and other sinks supported by collectd, telegraf, etc).

palitu, 2020-01-21:
What i was thinking that you basically write in the protocol you are using,
but deliver it to a broker, instead of directly.  We tend to use this to
allow for moving across firewalls easier, and aggregating numerous sites
into a central historian, without having to worry about a whole bunch of
things.

ie the message payload would be line protocol for influx, JSON for a
generic, and whatever the other ones are.

nickbabcock, 2020-01-22:
Seems like a good idea!

nickbabcock, 2020-11-25:
Closing this as this seems a bit out of date. Feel free leave a comment or open a new issue with the current status of desired changes

closed 2020-11-25 by nickbabcock