nickbabcock/OhmGraphite #118: On additional sensor libraries

nickbabcock, 2020-01-24:
LibrehardwareMonitor is good, but there may be other sensor libraries that expose more sensors or may be more complete. I will always back LibrehardwareMonitor as the default sensor library but it may prove useful to add additional integrations:

hwinfo allows one to export a csv of the data like so:

```
Date,Time,"Virtual Memory Commited [MB]","Virtual Memory Available [MB]","Virtual Memory Load [%]","Physical Memory Used [MB]","Physical Memory Available [MB]","Physical Memory Load [%]","Page File Usage [%]","Core #0 VID [V]","Core #1 VID [V]","Core #2 VID [V]","Core #3 VID [V]","Core #4 VID [V]","Core #5 VID [V]","Core #6 VID [V]","Core #7 VID [V]","Core #0 Clock (perf #8) [MHz]","Core #1 Clock (perf #6) [MHz]","Core #2 Clock (perf #7) [MHz]","Core #3 Clock (perf #5) [MHz]","Core #4 Clock (perf #2) [MHz]","Core #5 Clock (perf #3) [MHz]","Core #6 Clock (perf #1) [MHz]","Core #7 Clock (perf #4) [MHz]","CPU (Tctl/Tdie) [°C]","CPU Core Voltage (SVI2 TFN) [V]","SoC Voltage (SVI2 TFN) [V]","CPU Core Current (SVI2 TFN) [A]","SoC Current (SVI2 TFN) [A]","CPU Package Power (SMU) [W]","Core #0 Power (SMU) [W]","Core #1 Power (SMU) [W]","Core #2 Power (SMU) [W]","Core #3 Power (SMU) [W]","Core #4 Power (SMU) [W]","Core #5 Power (SMU) [W]","Core #6 Power (SMU) [W]","Core #7 Power (SMU) [W]","CPU Core Power (SVI2 TFN) [W]","SoC Power (SVI2 TFN) [W]","CPU+SoC Power (SVI2 TFN) [W]","Memory Controller Clock (UCLK) [MHz]","Motherboard [°C]","CPU [°C]","Auxiliary [°C]","AUXTIN1 [°C]","AUXTIN2 [°C]","CPU (PECI) [°C]","Vcore [V]","AVCC [V]","3VCC [V]","VIN4 [V]","3VSB [V]","VTT [V]","VIN5 [V]","VIN6 [V]","VIN2 [V]","VIN3 [V]","CPUFANIN0 [RPM]","Chassis Intrusion [Yes/No]","Drive Temperature [°C]","Drive Remaining Life [%]","Drive Failure [Yes/No]","Drive Warning [Yes/No]","Total Host Writes [GB]","Total Host Reads [GB]","Read Activity [%]","Write Activity [%]","Total Activity [%]","Read Rate [MB/s]","Write Rate [MB/s]","Read Total [MB]","Write Total [MB]","GPU Temperature [°C]","GPU Core Voltage [V]","GPU Fan [RPM]","GPU Power [W]","GPU Clock [MHz]","GPU Memory Clock [MHz]","GPU Video Clock [MHz]","GPU Core Load [%]","GPU Memory Controller Load [%]","GPU Video Engine Load [%]","GPU Bus Load [%]","GPU Memory Usage [%]","GPU D3D Usage [%]","GPU Video Decode 0 Usage [%]","GPU Video Encode 0 Usage [%]","GPU Video Encode 1 Usage [%]","GPU Computing (Compute_0) Usage [%]","GPU Computing (Compute_1) Usage [%]","GPU VR Usage [%]","GPU Fan [%]","Performance Limit - Power [Yes/No]","Performance Limit - Thermal [Yes/No]","Performance Limit - Reliability Voltage [Yes/No]","Performance Limit - Max Operating Voltage [Yes/No]","Performance Limit - Utilization [Yes/No]","Performance Limit - SLI GPUBoost Sync [Yes/No]","Total GPU Power (normalized) [% of TDP] [%]","Total GPU Power [% of TDP] [%]","GPU Memory Allocated [MB]","GPU D3D Memory Dedicated [MB]","GPU D3D Memory Dynamic [MB]","PCIe Link Speed [GT/s]","Total DL [MB]","Total UP [MB]","Current DL rate [KB/s]","Current UP rate [KB/s]","Total DL [MB]","Total UP [MB]","Current DL rate [KB/s]","Current UP rate [KB/s]","Total Errors []",
22.1.2020,5:27:6.575,27355,10209,72.8,20781,11919,63.5,4.3,1.100,1.025,1.075,1.087,1.138,1.150,1.131,1.188,3639.1,3391.6,3540.1,2360.1,2772.7,1639.4,1485.4,1386.3,42.0,1.019,0.912,10.541,9.118,25.681,0.449,1.137,0.799,0.519,0.519,0.560,0.512,0.811,10.739,8.320,19.059,1452.4,36.0,42.0,36.0,108.0,109.0,40.0,0.592,3.376,3.360,0.816,3.424,1.848,0.176,0.128,1.808,1.736,982,No,54,100.0,No,No,10305,3956,0.1,0.3,0.4,0.040,0.232,17088,51386,50,0.725,917,34.454,1531.0,2003.4,1366.5,1.0,1.0,0.0,0.0,12.9,2.0,0.0,0.0,0.0,0.0,0.0,0.0,40.0,No,No,No,No,Yes,No,29.4,22.8,1053,958,239,8.0,0,0,0.000,0.000,25,5,0.163,0.104,0,
```

The csv isn't perfect. The header `Core #3 Power (SMU) [W]` lacks the context that it's from my Ryzen 2700. This label is provided in hwinfo's UI. Currently OhmGraphite provides hardware labels to sensor readings, so ideally there would be a way to extract this.

hwinfo also has an SDK, but it appears to be under wraps (not open sourced / available)

[Aida64 allows external applications to access sensor readings.](https://www.aida64.co.uk/user-manual/external-applications) This is done via memmap. Aida64 is a paid-for application, right? So can't say my motivation is high.

mwasowski, 2020-05-04:
This [Remote Sensor Monitor](https://www.hwinfo.com/forum/threads/introducing-remote-sensor-monitor-a-restful-web-server.1025) is interesting and works, sadly not maintained for a long time.

Unlike csv report, it adds groups to the data [like so](https://imgur.com/cwIZ15G)

And I would prefer just to send data as OhmGraphite does instead of having to spin up web server in the background etc.

nickbabcock, 2020-05-23:
With OpenHardwareMonitor receiving recent updates, it is possible that Libre and Open could diverge. Maybe the first step to additional sensor libraries is to allow one to select Libre vs Open via the config file.

mwasowski, 2020-05-24:
I see, not bad idea.

nickbabcock, 2020-11-29:
Closing this as I don't see this becoming a priority any time soon.

closed 2020-11-29 by nickbabcock