nickbabcock/OhmGraphite #119: Feature request: SMART support

aaronjwood, 2020-02-02:
First of all, I'd like to say nice work on this tool! It complements telegraf on Windows very well and fills in a lot of gaps. It'd be awesome if the tool could monitor SMART similar to how you can do it with telegraf on Linux:
[image]

AFAIK there is nothing in telegraf for SMART on Windows. LibreHardwareMonitor seems to have decent support for reading SMART attributes.

nickbabcock, 2020-02-03:
Thanks for the kind words. I love this idea! For me LibreHardwareMonitor omits SMART attributes in the UI and includes it in the report.

Couple of speed bumps for the implementation: the SMART attributes are not exposed as sensors in LibreHardwareMonitor. And we'd need to reverse the non-standard lingo. For instance, in the report I see "Controller Busy Time", "Media Errors", etc and I don't believe these are standard smart attributes. Everything appears solvable, but may take a bit to integrate.

aaronjwood, 2020-02-22:
Cool, makes sense! Even if we started small like reporting healthy/unhealthy that'd be enough for me!

nickbabcock, 2020-02-23:
Yes, you're right. Something is better than nothing, and I should get a proof of concept out.

The one thing causing me programmer-hesitation is introducing sensors that are still being iterated on, as we add features or fix bugs. I don't want users getting flustered with a potentially high rate of churn for smart metrics. I was thinking perhaps adding a config `smart_sensors` that one can opt into, prefixing metric names with `beta_`, or maybe some documentation is all that is needed.

nickbabcock, 2020-02-28:
An update: looks like there are some SMART attributes exposed as sensors already, but not at all of them. I created a PR to add some: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/210 but it looks like only sensors should be in the desktop program will be exposed.

Back to some of my original concerns:

> reverse the non-standard lingo

Since some attributes are being exposed in the non-standard lingo already, there is no harm in adding to it.

> SMART attributes are not exposed as sensors 

I believe the fix will be to write our own pseudo-sensor that is populated when we detect a nvme harddrive

aaronjwood, 2020-03-01:
Do you know if LHW only supports this with nvme drives? Wondering if it has generic smart support or if it at least supports sata + nvme

nickbabcock, 2020-03-10:
Yes, LHW exposes SMART sensors for sata drives through the [ATAStorage](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/445e80159551b4b4c8ab4f2c8397b43a175b0210/LibreHardwareMonitorLib/Hardware/Storage/ATAStorage.cs) class and standard SMART attributes are exposed for a [generic hard disk](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/445e80159551b4b4c8ab4f2c8397b43a175b0210/LibreHardwareMonitorLib/Hardware/Storage/GenericHardDisk.cs). Let me know if you're not seeing SMART sensors from SATA drives. NVMe drives, on the other hand, don't inherit this chain of functionality, which partially leaves me to believe that there is some fundamental difference between SATA SMART and NVMe SMART. Looking at [NVMe 1.1 spec](https://nvmexpress.org/wp-content/uploads/NVM-Express-1_1b.pdf) (section 5.10.1.2 SMART / Health Information (Log Identifier 02h)), it does appear that it doesn't contain typical SATA SMART attributes. Perusing the CrystalDiskInfo's source code, it seems to try and find some middle ground by [combining NVMe SMART data and SATA SMART](https://github.com/hiyohiyo/CrystalDiskInfo/blob/8fc212774b47ae11323dcb73afbc4f593e1aec1a/NVMeInterpreter.cpp#L151) (and [here](https://github.com/hiyohiyo/CrystalDiskInfo/blob/dc3916e741bf20978d7dc80928f0162d90741b01/AtaSmart.cpp#L3022), which makes me question why LHW appears to expose all SATA SMART attributes but they don't want to expose all NVMe SMART data. I'm kicking myself for getting rid of all my traditional HDDs for testing 😅

aaronjwood, 2020-03-23:
I also don't have any hdds in my windows machine to test :) Only nvme for me!

nickbabcock, 2020-04-18:
I created #134 to address this issue. It doesn't expose all of the sensors, so let me know if there is one that you really desire that's missing. We can always add the rest down the road.

closed 2020-04-19 by nickbabcock

aaronjwood, 2020-04-19:
Thanks a ton for adding this. Unfortunately I don't see any related sensor in the release you just made:
```
2020-04-19 13:43:27.1724|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-04-19 13:43:27.1914|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|Starting as a Windows service
2020-04-19 13:43:27.1914|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Starting
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/0 "Fan Control #1"
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/1 "Fan Control #2"
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/2 "Fan Control #3"
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/3 "Fan Control #4"
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/4 "Fan Control #5"
2020-04-19 13:43:27.3244|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/control/5 "Fan Control #6"
2020-04-19 13:43:27.3894|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/1 "CPU Core #1"
2020-04-19 13:43:27.3894|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/2 "CPU Core #2"
2020-04-19 13:43:27.3894|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/3 "CPU Core #3"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/4 "CPU Core #4"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/5 "CPU Core #5"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/6 "CPU Core #6"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/7 "CPU Core #7"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/8 "CPU Core #8"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/9 "CPU Core #9"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/10 "CPU Core #10"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/11 "CPU Core #11"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/12 "CPU Core #12"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/13 "CPU Core #13"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/14 "CPU Core #14"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/15 "CPU Core #15"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/16 "CPU Core #16"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/17 "CPU Core #17"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/18 "CPU Core #18"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/19 "CPU Core #19"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/20 "CPU Core #20"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/21 "CPU Core #21"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/22 "CPU Core #22"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/23 "CPU Core #23"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/24 "CPU Core #24"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/25 "CPU Core #25"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/26 "CPU Core #26"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/27 "CPU Core #27"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/28 "CPU Core #28"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/29 "CPU Core #29"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/30 "CPU Core #30"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/31 "CPU Core #31"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/32 "CPU Core #32"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/0 "CPU Total"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/0 "Package Power"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/0 "Core #1"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/0 "Core #1"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/1 "Core #1 (SMU)"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/2 "Core #1 VID"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/1 "Core #2"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/1 "Core #2"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/2 "Core #2 (SMU)"
2020-04-19 13:43:27.3914|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/3 "Core #2 VID"
[... cut ...]
2020-04-19 13:43:40.1272|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/fan/1 "Fan #2"
2020-04-19 13:43:40.1272|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/fan/2 "Fan #3"
2020-04-19 13:43:40.1272|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/fan/3 "Fan #4"
2020-04-19 13:43:40.1272|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/fan/4 "Fan #5"
2020-04-19 13:43:40.1272|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6796dr/fan/5 "Fan #6"
```

nickbabcock, 2020-04-19:
What kind of hard drive do you have? Perchance is it a SATA m.2?

aaronjwood, 2020-04-20:
It's an HP EX950. NVMe via PCIe 3.0 x4

nickbabcock, 2020-04-20:
Yup that's definitely NVMe. I have a bunch of HP EX920's. Love'em and I assume the EX950 is even better. With my EX920, I see the following sensors on startup:

```
Hardware added: /nvme/0
Sensor added: /nvme/0/temperature/0 "Temperature"
Sensor added: /nvme/0/level/1 "Available Spare"
Sensor added: /nvme/0/level/2 "Available Spare Threshold"
Sensor added: /nvme/0/level/3 "Percentage Used"
Sensor added: /nvme/0/data/4 "Data Read"
Sensor added: /nvme/0/data/5 "Data Written"
Sensor added: /nvme/0/load/0 "Used Space"
Sensor added: /nvme/0/load/32 "Write Activity"
Sensor added: /nvme/0/load/33 "Total Activity"
Sensor added: /nvme/0/throughput/34 "Read Rate"
Sensor added: /nvme/0/throughput/35 "Write Rate"
Sensor added: /nvme/0/factor/media_errors "Media Errors"
Sensor added: /nvme/0/factor/power_cycles "Power Cycles"
Sensor added: /nvme/0/factor/error_info_log_entries "Error Info Log Entries"
Sensor added: /nvme/0/factor/unsafe_shutdowns "Unsafe Shutdowns"
```

Notice that "nvme" is the prefix, not "hdd" (thanks for including your log!).

Hmm I believe this to be an issue with the underlying sensor library, LibreHardwareMonitor. I perused the code a bit and saw a lot of paths that would have the drive fallback to a generic hdd implementation if there was an issue with nvme detection. I may look to log a diagnostic report under debug on startup so that we can gather more information.

aaronjwood, 2020-04-20:
No problem, and sounds good. If I get some time this week I'll do some breakpoint debugging on my system to see where it's going wrong.

aaronjwood, 2020-04-20:
FWIW I'm running on x570 using an ASRock Taichi.

aaronjwood, 2020-04-23:
Fixed with https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/232
If that gets merged soon I'll send you a PR with an update to the submodule.

aaronjwood, 2020-04-23:
Would you mind testing that patch with your drive/system?

nickbabcock, 2020-04-23:
Nice! Tested it out and saw no adverse behavior changes -- everything still the same 👍

PR #134 (2020-04-19): Add NVMe SMART Sensors
Since LibreHardwareMonitor doesn't expose them as sensors, we expose our
own sensors. This doesn't cover all of the attributes exposed by smart (like
power on hours, which doesn't have a corresponding sensor type as of now),
but I figure something is better than nothing (and we can expose more in the
future).

[image]

Closes #119