nickbabcock/OhmGraphite #123: Not able to see CCD temps on a Zen 2 chip (3950x)

aaronjwood, 2020-02-22:
The latest version of LHM shows me my CCD temps like so:
[image]
I'm not sure if this needs to be pulled out separately with some extra code or if the current .dll being used by this project doesn't support this but it would be super nice to be able report on the CCD temps. The only thing I can pull out right now is this:
[image]
AFAIK the CCD temps (Tdie) are the actual temps at the dies, the Tctl/Tdie that's reported now are the actual temps with an offset applied which is not at all what I think most people would want to see.

nickbabcock, 2020-02-22:
Yup, can confirm. Looks like CCD temps were added just a few days after the pinned version of LHM that OhmGraphite uses. Let me re-pin to the latest master, do a smoke test, and release a new version.

Thank you for the repot

aaronjwood, 2020-02-22:
Awesome! Thanks for the quick response!

nickbabcock, 2020-02-22:
This should be included in the recently released v0.12.0 version, let me know if it's not what you expect 👌

aaronjwood, 2020-02-22:
Works great, thanks!

closed 2020-02-22 by aaronjwood