nickbabcock/OhmGraphite #127: After influxdb down for 20min, reported Temperature values are nonsensical

kenstir, 2020-03-25:
My influxdb machine was off the air for 20m.  After it booted, the temperatures reported are silly, like 4,294,955 degrees C.

Here's what it looks like in my dashboard:
[image]

Here's how I got there:
* installed v0.12.0 and configured it for InfluxDB
* installed the sample dashboard https://grafana.com/grafana/dashboards/11601 (and fixed all the Temperature graphs that were labelled "Wattage")
* turned of my influx box, turned in on 20 min later

Sample influx query:
```
> SELECT mean("value") FROM "Temperature" WHERE ("app" = 'ohm' AND "hardware_type" =~ /^CPU$/ AND "host" =~ /^PEPE$/) AND time >= now() - 1h GROUP BY time(120s), "sensor" fill(null)

name: Temperature
tags: sensor=Core CCDs Average (Tdie)
time                 mean
----                 ----
2020-03-25T00:20:00Z
2020-03-25T00:22:00Z
2020-03-25T00:24:00Z
2020-03-25T00:26:00Z
2020-03-25T00:28:00Z
2020-03-25T00:30:00Z 2147497
2020-03-25T00:32:00Z 2147497.0833333335
2020-03-25T00:34:00Z 2684289.1666666665
2020-03-25T00:36:00Z 2863219.1666666665
2020-03-25T00:38:00Z 2863219
2020-03-25T00:40:00Z 2863219
2020-03-25T00:42:00Z 2863219
2020-03-25T00:44:00Z 2863219.1666666665
2020-03-25T00:46:00Z 2863219.3333333335
2020-03-25T00:48:00Z 2863220
2020-03-25T00:50:00Z 2863221.6666666665
2020-03-25T00:52:00Z 2863224.5
2020-03-25T00:54:00Z 2863224.25
2020-03-25T00:56:00Z 2863224.25
2020-03-25T00:58:00Z 2863224
2020-03-25T01:00:00Z 2863224.625

name: Temperature
tags: sensor=Core CCDs Max (Tdie)
time                 mean
----                 ----
2020-03-25T00:20:00Z
2020-03-25T00:22:00Z
2020-03-25T00:24:00Z
2020-03-25T00:26:00Z
2020-03-25T00:28:00Z
2020-03-25T00:30:00Z 4294955
2020-03-25T00:32:00Z 4294955
2020-03-25T00:34:00Z 4294955
2020-03-25T00:36:00Z 4294955
2020-03-25T00:38:00Z 4294955
2020-03-25T00:40:00Z 4294955
2020-03-25T00:42:00Z 4294955
2020-03-25T00:44:00Z 4294955
2020-03-25T00:46:00Z 4294955
2020-03-25T00:48:00Z 4294955
2020-03-25T00:50:00Z 4294955
2020-03-25T00:52:00Z 4294955
2020-03-25T00:54:00Z 4294955
2020-03-25T00:56:00Z 4294955
2020-03-25T00:58:00Z 4294955
2020-03-25T01:00:00Z 4294955
```

nickbabcock, 2020-03-25:
Thanks for dropping by, couple of questions:

- Is OhmGraphite and InfluxDB on the same or separate hosts? If on separate hosts, you saw this behavior only when InfluxDB was unavailable?
- Were you able to resolve the issue by restarting Influxdb or OhmGraphite?
- Is this issue reproducible (eg, making influxdb unavailable causes this issue every time)?
- This is with OhmGraphite v0.12, correct? I'm assuming you have a zen 2 cpu. Since v0.12 updated some zen 2 temperature logic, it may be worth downgrading to [v0.11](https://github.com/nickbabcock/OhmGraphite/releases/tag/v0.11.0) and seeing if the issue persists.

kenstir, 2020-03-25:
Thanks for the quick response and the suggestion!  

> Is OhmGraphite and InfluxDB on the same or separate hosts

OhmGraphite and InfluxDB are on different hosts.   I saw this issue only after InfluxDB came back online.

> Were you able to resolve the issue by restarting Influxdb or OhmGraphite?

I was able to resolve the issue by restarting the OhmGraphite service.  After the restart only 2 temps are visible, the rest went back to null.  See attached graph.

[image]

> Is this issue reproducible

This reproduces only sometimes.  It happened yesterday after I rebooted the InfluxDB host, and today after I shut down my work VPN.  It did not happen overnight when the desktop suspended.

> OhmGraphite v0.12, correct?  zen 2 cpu?

Yes, OhmGraphite v0.12.  I think Zen 2, it's a Ryzen 7 3700X.

nickbabcock, 2020-03-26:
Thanks for the info. I'm testing with a Ryzen 7 2700 right now (no zen 2 cpu on hand) against influxdb 1.7.9. I haven't been able to reproduce so far (stopping influxdb for awhile before restarting), but I'll keep trying. I have to figure out how the availability of influxdb could effect the sensor reading on the cpu -- that seems most bizarre to me 😅 

If you discover anymore clues let me know!

nickbabcock, 2020-04-19:
I released v0.13.0. This contained some fixes in the underlying LibreHardwareMonitor, so try it out and let me know if this issue persists.

kenstir, 2020-04-20:
Looks good so far!  2 of 2 times the monitor came back with sensible CPU temps.  They have different names than before but now we have a way to set that too.  Thank you!

nickbabcock, 2020-04-21:
Great to hear!

closed 2020-04-21 by nickbabcock