nickbabcock/OhmGraphite #130: Feature request: Allow metric names edit

mwasowski, 2020-04-16:
Hi there! Boy oh boy I am happy I've found your project :)

I started homelabbing not so long ago and after doing a bit of research I was actually giving up on an idea that I will be able to include my Windows workstation in Grafana dashboard I was slowly building for my server. But apparently I can, with the metrics I wanted (thermals, THERMALS!) :)

I have a small question/feature request. I know I can achieve this by changing display (crude way I've found so far was to split values I want to change to separate queries instead of doing GROUP BY tag(sensor) - not that efficient) but I would rather fix the data than the display, if possible that is.
So..I would like to have the ability to to set names for the sensors/metrics like it is possible in the LHM app by editing in the UI that gets stored in AppSettings in LibreHardwareMonitor.config

```
<add key="/amdcpu/0/load/1/name" value="CPU Core 0 T0" />
<add key="/amdcpu/0/load/2/name" value="CPU Core 0 T1" />
<add key="/amdcpu/0/load/3/name" value="CPU Core 1 T0" />
<add key="/amdcpu/0/load/4/name" value="CPU Core 1 T1" />
<add key="/amdcpu/0/load/5/name" value="CPU Core 2 T0" />
<add key="/amdcpu/0/load/6/name" value="CPU Core 2 T1" />
<add key="/amdcpu/0/load/7/name" value="CPU Core 3 T0" />
<add key="/amdcpu/0/load/8/name" value="CPU Core 3 T1" />
<add key="/amdcpu/0/load/9/name" value="CPU Core 4 T0" />
<add key="/amdcpu/0/load/10/name" value="CPU Core 4 T1" />
<add key="/amdcpu/0/load/11/name" value="CPU Core 5 T0" />
<add key="/amdcpu/0/load/12/name" value="CPU Core 5 T1" />
<add key="/amdcpu/0/load/13/name" value="CPU Core 6 T0" />
<add key="/amdcpu/0/load/14/name" value="CPU Core 6 T1" />
<add key="/amdcpu/0/load/15/name" value="CPU Core 7 T0" />
<add key="/amdcpu/0/load/16/name" value="CPU Core 7 T1" />
```

nickbabcock, 2020-04-16:
Welcome to the world of homelabbing, hope you enjoy your stay!

> So..I would like to have the ability to to set names for the sensors/metrics like it is possible in the LHM app by editing in the UI that gets stored in AppSettings in LibreHardwareMonitor.config

Wait, this is possible today in LHM? I had no idea, that's awesome. Can you link me somewhere (like docs or a comment or something) where it is explained as I don't see a `LibreHardwareMonitor.config` on my end.

kenstir, 2020-04-16:
+1 for this request.  I don't know where the docs are, but in the LHM GUI I just click the name I want to rename, e.g. "Fan #1", press F2 (windows), then type in the name I want, "HWM calls this SYSFANIN", and LHM saves it to LibreHardwareMonitor.config.

```
 <add key="/lpc/nct6798d/fan/0/name" value="HWM calls this SYSFANIN" />
```

nickbabcock, 2020-04-17:
Oh wow, I can't believe I wasn't aware of this feature! Don't sweat trying to find docs then I was just looking for a way to replicate it.

I was having second thoughts if I should introduce an additional config for those who don't like xml, but being able to copy paste between LibreHardwareMonitor and OhmGraphite would be a huge boon. I'll see what I can do

nickbabcock, 2020-04-17:
Alright, so I created #131 for this feature. I'll wait a little bit to see if any problems shake out.

mwasowski, 2020-04-17:
Fantastic! Sorry for not replying earlier and not including it in my original post. It's as @kenstir said, you need to first make a change to the config file to be generated for you.

I was initially thinking about making it so the config gets autogenerated like when you edit something, so that copy&paste is not required. But that also is an issue for someone that does not want to use aliasing. So maybe a property `UseCustomSensorAliasNames` or something. But that's also another level of complexity so I am completely fine with copy&paste! :)

nickbabcock, 2020-04-17:
> I was initially thinking about making it so the config gets autogenerated like when you edit something, so that copy&paste is not required

Just so we're talking about the same thing, are you suggesting the OhmGraphite modifies it's own config on startup to include all default sensor names?

Or are you talking about OhmGraphite detecting alias entries from a LibreHardwareMonitor.config (like the LibreHardwareMonitor.config is stored in the same directory or pointed at by a config field)?

mwasowski, 2020-04-17:
At first I was thinking about having another config, because I thought that OhmGraphite config is json not xml. The fact that it's xml, makes me not want to have another file, that would be just confusing because we could just have a section in OhmGraphite.config.

And as for your question I was thinking about the first option, dump sensors to the config on start and use it or not, your call as a user. Or control it with a setting in main app settings config file, but that might be a bad idea from the design perspective, not sure.

We could start with that, but I definitely feel that the need to download LHM, rename sensor, copy config entry from it's config to OhmGraphite config would be quite cumbersome in a long run. Though in most cases it's one time action, most of this stuff does not change.

nickbabcock, 2020-04-17:
Another way to tease out the sensor name to id relationship is the `OhmGraphite.log`. Under debug mode it will log the sensor id

```
Sensor added: /lpc/nct6792d/control/0
```

Perhaps the log can be changed to write these messages under info (ie: the default) and include the sensor name as well

```
Sensor added: /lpc/nct6792d/control/0 "Fan Control #1"
```

Thoughts on that?

mwasowski, 2020-04-17:
I like that idea. It does involve manual actions, but at the same time, we'll have all the information we need. I was thinking that maybe separate log for that but that might be just unnecessary complications since all this I expect to be at the start of the service, so technically one place, not like one here one there?

nickbabcock, 2020-04-17:
Yeah if someone wants to find the sensor id -> name mapping then I'm fine with them either configuring via LibreHardwareMonitor or searching for it in the OhmGraphite.log

closed 2020-04-17 by nickbabcock