nickbabcock/OhmGraphite #139: No Connection

LikeTheYogurt, 2020-05-02:
Hi, 

I have tried to follow the steps to get this up and running but i am seeing this error:
2020-05-01 23:42:03.9399|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-05-01 23:42:03.9555|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-05-01 23:42:03.9555|ERROR|Topshelf.Hosts.ConsoleRunHost|The OhmGraphite service is running and must be stopped before running via the console
2020-05-01 23:42:05.6521|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.Net.Sockets.SocketException (0x80004005): No connection could be made because the target machine actively refused it 127.0.0.1:2003
   at System.Net.Sockets.Socket.InternalEndConnect(IAsyncResult asyncResult)
   at System.Net.Sockets.Socket.EndConnect(IAsyncResult asyncResult)
   at System.Net.Sockets.TcpClient.EndConnect(IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.GraphiteWriter.<ReportMetrics>d__9.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()

Any idea as to what i need to do?

nickbabcock, 2020-05-02:
The crux of the problem can be seen in the log:

> No connection could be made because the target machine actively refused it 127.0.0.1:2003

OhmGraphite, by default, is setup to write to the Graphite database on 127.0.0.1:2003. Is this where your graphite server is running? If not, you'll need to update the config with the correct host and port.

LikeTheYogurt, 2020-05-02:
Ah! I see. I am wanting to use Prometheus. 

So i need to update the config file to the Prometheus host and port?

When i do that, I see this in the log: 

2020-05-02 16:13:03.8989|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.IO.IOException: Unable to write data to the transport connection: An existing connection was forcibly closed by the remote host. ---> System.Net.Sockets.SocketException: An existing connection was forcibly closed by the remote host
   at System.Net.Sockets.Socket.BeginSend(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags, AsyncCallback callback, Object state)
   at System.Net.Sockets.NetworkStream.BeginWrite(Byte[] buffer, Int32 offset, Int32 size, AsyncCallback callback, Object state)
   --- End of inner exception stack trace ---
   at System.Net.Sockets.NetworkStream.BeginWrite(Byte[] buffer, Int32 offset, Int32 size, AsyncCallback callback, Object state)
   at System.IO.Stream.<>c.<BeginEndWriteAsync>b__53_0(Stream stream, ReadWriteParameters args, AsyncCallback callback, Object state)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncTrim[TInstance,TArgs](TInstance thisRef, TArgs args, Func`5 beginMethod, Func`3 endMethod)
   at System.IO.Stream.BeginEndWriteAsync(Byte[] buffer, Int32 offset, Int32 count)
   at System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count, CancellationToken cancellationToken)
   at System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count)
   at System.IO.StreamWriter.<FlushAsyncInternal>d__68.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.IO.StreamWriter.<WriteAsyncInternal>d__55.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at OhmGraphite.GraphiteWriter.<ReportMetrics>d__9.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()

nickbabcock, 2020-05-02:
Still looks like you have OhmGraphite writing to Graphite. Check out the [Prometheus Configuration docs](https://github.com/nickbabcock/OhmGraphite/tree/9c506ba50d8e37e6eaf1c2cb6a207a34603437f4#prometheus-configuration)

LikeTheYogurt, 2020-05-05:
Thanks for the help Nick and thanks for the useful tool!

closed 2020-05-05 by LikeTheYogurt