nickbabcock/OhmGraphite #140: Feature request: Allow metric hiding

mwasowski, 2020-05-04:
Hi Nick!

Great work on adding the aliasing in v0.13.0, works like a charm! I have another one for you that is kind of a followup to my previous request.

It would be great to add ability to hide specific metrics and not log them. OHM supports this type of config entries that make then not appear on the list unless show hidden is turned on:
```
<add key="/lpc/it8665e/temperature/0/hidden" value="true" />
```
OhmGraphite could skip entries with `hidden = true` so we're not sending data we do not want/do not need, or simply because OHM values are incorrect and therefore bring no value to the monitoring.

For example, on my Asus Crosshair Hero VII [those temps](https://imgur.com/RMGTsDe) are useless because they are just duplicated CPU Socket Temperatures.

nickbabcock, 2020-05-06:
Agreed, this sounds like a good idea.

mwasowski, 2020-05-06:
Perfect!

closed 2020-05-23 by nickbabcock

nickbabcock, 2020-06-05:
This has been released as part of v0.15.0

mwasowski, 2020-06-08:
Did I say already you're awesome? :D Anyway, you are!

PR #144 (2020-05-23): Hide sensors from export with configuration
There may be a sensor that is faulty on a given machine. Maybe it reports negative temperatures. This can throw off monitoring software or make it harder to maintain with all the special cases. OhmGraphite allows one to exclude a sensor from being exported by modifying the OhmGraphite config and adding the `/hidden` suffix to the sensor id like so:

```xml
<add key="/lpc/nct6792d/temperature/1/hidden" />
```

Closes #140