nickbabcock/OhmGraphite #143: Not able to see CPU temps on a Zen 2 chip (3950x)

FLeven, 2020-05-23:


nickbabcock, 2020-05-23:
Thanks for the bug report! Couple questions:

- Which version of OHM worked? Do you happen to know which version of OHM started exposing coffee lake refresh (eg: was it 0.9.0 or 0.9.1, etc)?
- [Does the latest build of LibreHardwareMonitor](https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts) expose the i9-9880h?

FLeven, 2020-05-24:
Only tested the latest OHM Version and the latest LHM also shows the temperatures

FLeven, 2020-05-24:
OK, seems to be a OS related problem, when I start a different Windows 10 on my machine, all values are working and displayed unser metrics. CPU Temp/Core are not reported on the other Windows 10, maybe there is a service deactivated or some Lenovo Drivers are missing.

closed 2020-05-24 by FLeven