nickbabcock/OhmGraphite #145: Unable to send metrics error

acarbonaro, 2020-06-01:
When running `OhmGraphite` using the following command, the process starts and seems to add sensors, but loops through an exception.

`C:\Monitoring\OhmGraphite-0.14.0\OhmGraphite.exe run`

I can run `LibreHardwareMonitor` as a GUI and see all the proper data.

Here's the exception:
```
2020-06-01 12:53:44.0342|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.ArgumentNullException: Value cannot be null.
Parameter name: stringToEscape
   at System.Uri.EscapeDataString(String stringToEscape)
   at InfluxDB.LineProtocol.Client.LineProtocolClient.<OnSendAsync>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.InfluxWriter.<ReportMetrics>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()
```

And here's the full output from `OhmGraphite.exe`:
```
PS C:\Monitoring\OhmGraphite-0.14.0> C:\Monitoring\OhmGraphite-0.14.0\OhmGraphite.exe run
2020-06-01 13:49:47.6166|INFO|OhmGraphite.Program|Influxdb address: http://192.168.1.29:8086/ db: gaming
2020-06-01 13:49:47.6825|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-06-01 13:49:47.6893|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/0 "Fan Control #1"
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/1 "Fan Control #2"
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/2 "Fan Control #3"
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/3 "Fan Control #4"
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/4 "Fan Control #5"
2020-06-01 13:49:47.7892|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6791d/control/5 "Fan Control #6"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/1 "CPU Core #1"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #2"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #3"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #4"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/0 "CPU Total"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Core #3"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #4"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/4 "CPU Package"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/5 "CPU Core #1 Distance to TjMax"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/6 "CPU Core #2 Distance to TjMax"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/7 "CPU Core #3 Distance to TjMax"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/8 "CPU Core #4 Distance to TjMax"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/9 "Core Max"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/10 "Core Average"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/1 "CPU Core #1"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/2 "CPU Core #2"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/3 "CPU Core #3"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/4 "CPU Core #4"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/0 "CPU Package"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/1 "CPU Cores"
2020-06-01 13:49:47.8466|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/2 "CPU Graphics"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/3 "CPU Memory"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/0 "Bus Speed"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/0 "Memory Used"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/1 "Memory Available"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/0 "Memory"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/2 "Virtual Memory Used"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/3 "Virtual Memory Available"
2020-06-01 13:49:47.8599|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/1 "Virtual Memory"
2020-06-01 13:49:47.9576|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu/0/temperature/0 "GPU Core"
[... 105 lines cut ...]
   at System.Uri.EscapeDataString(String stringToEscape)
   at InfluxDB.LineProtocol.Client.LineProtocolClient.<OnSendAsync>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.InfluxWriter.<ReportMetrics>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()
2020-06-01 13:49:58.8290|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.ArgumentNullException: Value cannot be null.
Parameter name: stringToEscape
   at System.Uri.EscapeDataString(String stringToEscape)
   at InfluxDB.LineProtocol.Client.LineProtocolClient.<OnSendAsync>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.InfluxWriter.<ReportMetrics>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()
2020-06-01 13:49:59.4190|INFO|Topshelf.Hosts.ConsoleRunHost|Control+C detected, attempting to stop service.
2020-06-01 13:49:59.4938|INFO|Topshelf.Hosts.ConsoleRunHost|The OhmGraphite service has stopped.
```

nickbabcock, 2020-06-01:
Post your OhmGraphite.exe.config: this is mine for influxdb and I can confirm that it is working:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="influxdb" />
    <add key="interval" value="5" />
    <add key="influx_address" value="http://10.0.0.201:8086" />
    <add key="influx_db" value="mydb" />
   </appSettings>
</configuration>
```

My best guess is that the username is specified without a password.

acarbonaro, 2020-06-01:
Ah, that was it.  The user is passwordless, but the password key is necessary if there's a user.  Thank you!

closed 2020-06-01 by acarbonaro