nickbabcock/OhmGraphite #153: Service Install Not Reading All Sensors

DDxPlague, 2020-08-12:
Hello,

If i run the executable directly from power shell all of my sensors are picked up and log into influx and Grafana. However if i install as a service, some of the sensors like GPU activity and temp stop getting logged. 

Is there a way to install the service and run it as an admin? I looked but couldn’t figure it out last night.

nickbabcock, 2020-08-12:
Perchance do you have an AMD GPU? Some have reported that their AMD GPU sensors disappear on wake. This shouldn't effect any OhmGaphite users, but if this is occurring for you, I can upload a tentative release here to see if it helps.

If not, then we'll have to do some more digging, as it sounds like you're doing everything correct.

Do the GPU sensors initially show up as a service and then are removed or never show up at all? If the latter, there should be a log: `OhmGraphite.log, it'll contain lines like:

```
2020-08-12 15:54:07.9393|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/temperature/0 "GPU Core"
2020-08-12 15:54:07.9393|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/0 "GPU Core"
2020-08-12 15:54:07.9393|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/1 "GPU Memory"
2020-08-12 15:54:07.9453|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/2 "GPU Shader"
```

Compare the sensors that are logged when executing `OhmGraphite.exe run` and `OhmGraphite.exe start`. If you see all the GPU sensors being logged in both, then...I may need to think a bit more on that.

Also search the log for any "ERROR" lines.

DDxPlague, 2020-08-12:
I do have an AMD GPU. It's an RX590 if that helps.

Last night while I was debugging I noticed that the sensors for the GPU showed up in the logs when I ran the `OhmGraphite.exe run` command, but were not logged when the service was started.

nickbabcock, 2020-08-12:
See attached. Try it out and see if your issue is fixed. 🤞 

[OhmGraphite-0.15.1.zip](https://github.com/nickbabcock/OhmGraphite/files/5066394/OhmGraphite-0.15.1.zip)

DDxPlague, 2020-08-13:
Sadly no dice still. :( 

Log when started from Powershell:

DDxPlague, 2020-08-13:
Logs when started from Service:

```
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-08-12 20:09:45.9501|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-08-12 20:09:45.9666|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|Starting as a Windows service
2020-08-12 20:09:45.9666|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Starting
2020-08-12 20:09:46.0598|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/0 "Fan Control #1"
2020-08-12 20:09:46.0598|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/1 "Fan Control #2"
2020-08-12 20:09:46.0598|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/2 "Fan Control #3"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/1 "CPU Core #1"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #2"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #3"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #4"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/0 "CPU Total"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Core #3"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #4"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/4 "CPU Package"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/5 "CPU Core #1 Distance to TjMax"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/6 "CPU Core #2 Distance to TjMax"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/7 "CPU Core #3 Distance to TjMax"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/8 "CPU Core #4 Distance to TjMax"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/9 "Core Max"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/10 "Core Average"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/1 "CPU Core #1"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/2 "CPU Core #2"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/3 "CPU Core #3"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/4 "CPU Core #4"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/0 "CPU Package"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/1 "CPU Cores"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/2 "CPU Graphics"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/0 "Bus Speed"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/0 "Memory Used"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/1 "Memory Available"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/0 "Memory"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/2 "Virtual Memory Used"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/3 "Virtual Memory Available"
2020-08-12 20:09:46.1197|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/1 "Virtual Memory"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/temperature/0 "Temperature"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/data/0 "Total Bytes Written"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/load/0 "Used Space"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/load/32 "Write Activity"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/load/33 "Total Activity"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/throughput/34 "Read Rate"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /ssd/0/throughput/35 "Write Rate"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/temperature/0 "Temperature"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/0 "Used Space"
2020-08-12 20:09:46.7954|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/32 "Write Activity"
2020-08-12 20:09:46.8064|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/33 "Total Activity"
2020-08-12 20:09:46.8064|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/throughput/34 "Read Rate"
[... cut ...]
2020-08-12 20:09:52.1347|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/fan/0 "Fan #1"
2020-08-12 20:09:52.1347|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/fan/1 "Fan #2"
2020-08-12 20:09:52.1347|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/fan/2 "Fan #3"
2020-08-12 20:09:52.1347|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/fan/3 "Fan #4"
2020-08-12 20:09:52.1347|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/fan/4 "Fan #5"
2020-08-12 20:09:52.4554|INFO|OhmGraphite.MetricTimer|Sent 108 metrics in 348.7251ms
```

nickbabcock, 2020-08-13:
Thank you for the logs! They very clearly show the 8 missing GPU sensors when running as a service. I'm not sure why this would be. Some things that I'm curious about:

- Could there be a difference in the user running the service vs running it interactively? Maybe in your case there should (or shouldn't) be a different account running it.

[image]

- Maybe the `%PATH%` is significantly different between interactive and the service. Try locating `atiadlxx.dll` somewhere on your system and copying it to the folder where OhmGraphite is installed
- And then maybe as a sanity check, I'm curious if running LibreHardwareMonitor (https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts) will show the GPU (I'm assuming yes, but it is something to try)

DDxPlague, 2020-08-15:
LibreHardwareMonitor shows the GPU temps correctly.

I moved the `atiadlxx.dll` into the same directory but that didn't solve it.

I also tried running it as the same user as I am in Powershell, but it's still not pulling GPU sensors.

Any thoughts on where to go from here?

nickbabcock, 2020-08-15:
I talked to some people who are smarter than me and one of the suggestions is to try checking "Allow service to interact with desktop"

[image]

Seems like the root cause is that (for a yet unknown reason) windows is blocking access to the GPU from window services. Looks like plenty of people have this issue:

- https://github.com/hirschmann/nbfc/issues/203
- https://github.com/openhardwaremonitor/openhardwaremonitor/issues/833
- https://stackoverflow.com/questions/21267327/opengl-rendering-from-windows-service
- https://www.mathworks.com/matlabcentral/answers/98712-why-am-i-not-able-to-access-a-gpu-card-on-a-windows-vista-windows-7-machine-managed-under-job-manage

I do not know why GPU sensors for me and not for you (you'd think that if windows restricted GPU access from services that I wouldn't be able to see GPU sensor data either) 🤔 

Let me know if this works!

DDxPlague, 2020-08-18:
Sadly no dice on clicking the check box either. :(

nickbabcock, 2020-08-19:
Hmm, I'm running out of ideas. What version of Windows do you happen to be running?

DDxPlague, 2020-08-24:
I'm running Version	10.0.18363 Build 18363

nickbabcock, 2020-08-25:
Hmm yeah that is a recent build of windows. I'll leave this open for now, but I don't have any further steps for you. It may just boil down to the AMD's display driver being inaccessible to session 0 .

DDxPlague, 2020-10-09:
I was able to get this working with a workaround. :)  Slightly annoying as I have to click "okay" every time my computer starts, but it works and I don't have gaps now.

Workaround:

Start a scheduled task every time my user logs in. The actual scheduled task is:
Program/script: ```PowerShell.exe```
Add arguments (optional): ```-ExecutionPolicy Bypass -File C:\Users\tjeag\StartOhmGraphiteOnLogin.ps1```

And the contents of `C:\Users\tjeag\StartOhmGraphiteOnLogin.ps1`

```Start-Process -FilePath C:\OhmGraphite-0.15.0\OhmGraphite.exe run -Verb RunAs```

nickbabcock, 2020-10-10:
Oh wow that's an amazing idea!

I wonder if it's possible to alternatively place a `.bat` or powershell script in `shell:startup` that delegates to `OhmGraphite.exe`. I don't *think* it'd behave differently though sometimes the nuances of windows internals is lost on me.

Keeps us informed with whatever you discover! I'll add it to the docs so that future users don't have to struggle.

roy-spark, 2023-08-03:
Not only wrong GPU memory.

This (surge) is when switching from Service to direct run-command:

[image]

I also notice that seemingly it reports correctly in service mode on another machine that has another configuration. So this error will be "fun" to identify.