nickbabcock/OhmGraphite #157: Error: "The format of the specified network name is invalid"

gmariani, 2020-08-25:
I am trying to use Prometheus as the endpoint/server and have configured the file as such (i have the actual IP in my real file):

```
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="4445" />
    <add key="prometheus_host" value="255.255.255.255" />
  </appSettings>
</configuration>
```
But when I go to run ohmgraphite.exe i get this error message:
```
2020-08-25 15:03:57.8483|ERROR|OhmGraphite.PrometheusServer|Exception: starting prometheus server!System.Net.HttpListenerException (0x80004005): The format of the specified network name is invalid
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
2020-08-25 15:03:57.8640|ERROR|OhmGraphite.Program|OhmGraphite TopShelf encountered an error!System.Exception: Exception with starting prometheus server ---> System.Net.HttpListenerException: The format of the specified network name is invalid
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- End of inner exception stack trace ---
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   at Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   at Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   at Topshelf.Hosts.ConsoleRunHost.Run()
2020-08-25 15:03:57.8640|ERROR|Topshelf.Hosts.ConsoleRunHost|An exception occurred!System.Exception: Exception with starting prometheus server ---> System.Net.HttpListenerException: The format of the specified network name is invalid
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- End of inner exception stack trace ---
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   at Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   at Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   at Topshelf.Hosts.ConsoleRunHost.Run()
```

Am i only able to use a domain or am I not entering the IP correctly? If i switch it to `key="prometheus_host" value="*"` it works. If i switch it to `key="prometheus_host" value="example.com"` it will start. But I'd prefer to use an IP, any help would be appreciated.

nickbabcock, 2020-08-25:
Assuming the prometheus server is on another machine, the `prometheus_host` config field is the address that OhmGraphite listens on, it's not the prometheus server's address. `prometheus_host` is the address that OhmGraphite will bind to for prometheus to access. That is how "*" work -- `*` causes OhmGraphite to bind to all interfaces. It's...confusing and I welcome any suggestion to make the distinction more clear.

You should be able to find the appropriate `prometheus_host` by executing `Get-NetIPAddress | Format-Table` in powershell on the OhmGraphite host.

gmariani, 2020-08-27:
Ah ok, i think i get it now. I'm familiar with programs binding to ports, I didn't consider it could bind to a hostname. Prometheus itself doesn't send a 'host' header when it pings the metrics URL so it gets blocked. I checked on Prometheus' issue list and it doesn't look like that is something they will ever support https://github.com/prometheus/prometheus/issues/1724 (custom headers). So based on that, my only option here is to bind to * and call it a day? Is there any configuration in OhmGraphite I'm overlooking that would allow me to lock it down to a specific IP?

nickbabcock, 2020-08-28:
If you'd want to restrict which clients can connect to OhmGraphite (based on the client IP) then you'd need to use some sort of firewall unfortunately.

gmariani, 2020-08-30:
I agree, I came to that conclusion as well. Just wanted to know if there was a built-in method before going that route. Thanks.

closed 2020-08-30 by gmariani