nickbabcock/OhmGraphite #158: When use Prometheus ,OhmGraphite.MetricTimer|Unable to send metrics!System.IO.IOException

Fly0905, 2020-08-28:
I  use Prometheus.

So i according to  [the Prometheus Configuration](https://github.com/nickbabcock/OhmGraphite/tree/9c506ba50d8e37e6eaf1c2cb6a207a34603437f4#prometheus-configuration) docs update the config file to the Prometheus host and port.
`<?xml version="1.0" encoding="utf-8"?>
<configuration>
  <appSettings>
    <add key="host" value="localhost" />
    <add key="port" value="9090" />
    <add key="interval" value="5" />
  </appSettings>
  <startup>
    <supportedRuntime version="v4.0" sku=".NETFramework,Version=v4.6.1" />
  </startup>
</configuration>`
But when i do that, I see this in the log:
`2020-08-28 13:46:23.1390|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.IO.IOException: 无法将数据写入传输连接: 远程主机强迫关闭了一个现有的连接。。 ---> System.Net.Sockets.SocketException: 远程主机强迫关闭了一个现有的连接。
   在 System.Net.Sockets.Socket.BeginSend(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags, AsyncCallback callback, Object state)
   在 System.Net.Sockets.NetworkStream.BeginWrite(Byte[] buffer, Int32 offset, Int32 size, AsyncCallback callback, Object state)
   --- 内部异常堆栈跟踪的结尾 ---
   在 System.Net.Sockets.NetworkStream.BeginWrite(Byte[] buffer, Int32 offset, Int32 size, AsyncCallback callback, Object state)
   在 System.IO.Stream.<>c.<BeginEndWriteAsync>b__53_0(Stream stream, ReadWriteParameters args, AsyncCallback callback, Object state)
   在 System.Threading.Tasks.TaskFactory`1.FromAsyncTrim[TInstance,TArgs](TInstance thisRef, TArgs args, Func`5 beginMethod, Func`3 endMethod)
   在 System.IO.Stream.BeginEndWriteAsync(Byte[] buffer, Int32 offset, Int32 count)
   在 System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count, CancellationToken cancellationToken)
   在 System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count)
   在 System.IO.StreamWriter.<FlushAsyncInternal>d__68.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   在 System.IO.StreamWriter.<WriteAsyncInternal>d__55.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   在 OhmGraphite.GraphiteWriter.<ReportMetrics>d__9.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 OhmGraphite.MetricTimer.<ReportMetrics>d__7.MoveNext()
2020-08-28 13:46:24.3130|INFO|OhmGraphite.Program|Graphite host: localhost port: 9090 interval: 5 tags: False
2020-08-28 13:46:24.4157|INFO|Topshelf.HostFactory|Configuration Result:`

nickbabcock, 2020-08-28:
Are you missing:

```xml
    <add key="type" value="prometheus" />
```

from the config?  I don't see it in the code snippet you've posted.

Fly0905, 2020-08-31:
> Are you missing:
> 
> ```
>     <add key="type" value="prometheus" />
> ```
> 
> from the config? I don't see it in the code snippet you've posted.

Thanks，I add the config :
`<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="9090" />
    
    
    <add key="prometheus_host" value="10.0.50.162" />
  </appSettings>
</configuration>`
But when i do that, I see this in the log:
 2020-08-31 10:29:22.6144|INFO|OhmGraphite.Program|Prometheus port: 9090
2020-08-31 10:29:22.7656|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-08-31 10:29:22.7836|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.0.194, .NET Framework v4.0.30319.42000
2020-08-31 10:29:24.1295|ERROR|OhmGraphite.PrometheusServer|Exception: starting prometheus server!System.Net.HttpListenerException (0x80004005): 另一个程序正在使用此文件，进程无法访问。
   在 System.Net.HttpListener.AddAllPrefixes()
   在 System.Net.HttpListener.Start()
   在 Prometheus.MetricServer.StartServer(CancellationToken cancel)
   在 Prometheus.MetricHandler.Start()
   在 OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
2020-08-31 10:29:24.1454|ERROR|OhmGraphite.Program|OhmGraphite TopShelf encountered an error!System.Exception: Exception with starting prometheus server ---> System.Net.HttpListenerException: 另一个程序正在使用此文件，进程无法访问。
   在 System.Net.HttpListener.AddAllPrefixes()
   在 System.Net.HttpListener.Start()
   在 Prometheus.MetricServer.StartServer(CancellationToken cancel)
   在 Prometheus.MetricHandler.Start()
   在 OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- 内部异常堆栈跟踪的结尾 ---
   在 OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   在 Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   在 Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   在 Topshelf.Hosts.ConsoleRunHost.Run()
2020-08-31 10:29:24.1454|ERROR|Topshelf.Hosts.ConsoleRunHost|An exception occurred!System.Exception: Exception with starting prometheus server ---> System.Net.HttpListenerException: 另一个程序正在使用此文件，进程无法访问。
   在 System.Net.HttpListener.AddAllPrefixes()
   在 System.Net.HttpListener.Start()
   在 Prometheus.MetricServer.StartServer(CancellationToken cancel)
   在 Prometheus.MetricHandler.Start()
   在 OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- 内部异常堆栈跟踪的结尾 ---
   在 OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   在 Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   在 Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   在 Topshelf.Hosts.ConsoleRunHost.Run()

nickbabcock, 2020-08-31:
Couple things to try:

Load up the prometheus UI and see if you see OhmGraphite metrics like the following:

[image]

If you don't see anything you can verify that prometheus is able to access OhmGraphite by updating `NLog.config` and changing the logger levels to "Debug" (from "Info"):

```xml
  <rules>
    <logger name="*" minlevel="Debug" writeTo="file" />
    <logger name="*" minlevel="Debug" writeTo="console" />
  </rules>
``` 

When that is done you should see log lines like:

```
OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
```

If you don't see lines like that that means there is a connection / firewall issue, which you should double check the prometheus instructions: https://github.com/nickbabcock/OhmGraphite#prometheus-configuration

closed 2020-09-07 by nickbabcock