nickbabcock/OhmGraphite #159: [help required] Volts and FAN RPM Not Visible

vishallanke, 2020-08-28:
I have set up the utility and configured user provided Grafana dashboard for InfluxDB.
I can see that all the parameters like Temperature,  Frequencies are properly visible.
I can not see any values for Volts and FAN RPM.
Do I need to do any additional configuration ?

nickbabcock, 2020-08-28:
What is your motherboard? Not all motherboards expose these sensors

vishallanke, 2020-08-28:
I am using HP Elitebook.
I can see BaseBoard as HP
Baseboard Model - Not Specified

nickbabcock, 2020-08-29:
Yeah some OEM motherboards are poorly supported. You can try [OpenHardwareMonitor](https://openhardwaremonitor.org/) and [LibreHardwareMonitor](https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts) to see if they report it -- else there is not much I can do.

closed 2020-09-07 by nickbabcock