nickbabcock/OhmGraphite #160: RX 5700 Not Ingesting to InfluxDB

CHIEFDAC, 2020-09-14:
This might be a similar issue to ticket #153 but figured I would open a separate issue just in case.

All but 1 of my GPU's are appearing in Grafana, which is the RX 5700 mention in the title. I can confirm that HWInfo is logging the temperature data from this GPU:

[image]

Here is the log file for the host in question:

[OhmGraphite.5.log](https://github.com/nickbabcock/OhmGraphite/files/5215404/OhmGraphite.5.log)

Short snip-it of the logs for 4/5 of the GPU's:

2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/temperature/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/fan/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/control/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/clock/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/clock/2 "GPU Memory"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/voltage/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/0/load/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/power/3 "GPU Package"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/temperature/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/fan/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/control/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/clock/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/clock/2 "GPU Memory"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/voltage/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/8/load/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/power/3 "GPU Package"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/temperature/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/fan/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/control/0 "GPU Fan"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/clock/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/clock/2 "GPU Memory"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/voltage/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/16/load/0 "GPU Core"
2020-09-07 11:27:46.0958|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/power/3 "GPU Package"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/temperature/0 "GPU Core"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/fan/0 "GPU Fan"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/control/0 "GPU Fan"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/clock/0 "GPU Core"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/clock/2 "GPU Memory"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/voltage/0 "GPU Core"
2020-09-07 11:27:46.1099|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-amd/24/load/0 "GPU Core"

nickbabcock, 2020-09-14:
#153 is about the discrepancy in detected sensors when running:

- `OhmGraphite.exe start` (starts the service)
- `OhmGraphite.exe run` (run the program interactively in the terminal)

If the RX 5700 is missing in both than this is a different issue. These cards should be supported (see https://github.com/nickbabcock/OhmGraphite/issues/108) and since you already have other AMD cards, it's peculiar why one wouldn't be showing up 🤔 

Similar to HWiNFO, I'm curious if you can [try the latest build of LibreHardwareMonitor](https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts) and see if that shows your 5700.

CHIEFDAC, 2020-09-15:
This is strange... I was able to install LibreHardwareMonitor, but I'm seeing the same issue on that app like OhmGraphite:

**Note:** I've hidden the other items (e.g. Motherboard, Ram, HDD, etc.) to simplify the view; the monitoring information appeared correctly for the other hardware. I also enabled the option "View Hidden Sensors" and still have the same results.

[image]

nickbabcock, 2020-09-15:
Yeah it's a bummer that the [HWiNFO SDK is a "commercial product and [the pricing] is too high for small personal projects"](https://www.hwinfo.com/forum/threads/how-do-i-purchase-the-sdk.6126/post-23346). So I imagine that it would be more accurate than LibreHardwareMonitor / OpenHardwareMonitor .

CHIEFDAC, 2020-09-15:
Not a huge deal - OhmGraphite is still monitoring 5/6 on this host and it's monitoring both Radeon VII's on the other build I have :).

Let me know if you need any additional logs or information to assist with this project!

nickbabcock, 2020-09-16:
If you're feeling intrepid you can create a tracking issue over at the LibreHardwareMonitor and if there is a fix, it will be down streamed here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

CHIEFDAC, 2020-09-19:
Done - https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/302

nickbabcock, 2024-01-29:
I'm going to close this out as it looks like others who run 5700 have working setups: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/717

Let me know if otherwise.

closed 2024-01-29 by nickbabcock