nickbabcock/OhmGraphite #162: Unable to startup on Windows 10 professional

nwilging, 2020-09-25:
Awesome plugin, this _should_ do exactly what I need (send hardware temperature data to Influx).

The initial setup docs are very straightforward and I got going almost immediately after downloading the app. However, upon first startup, the metric-reporting process errors out. It is unable to send any metrics at all.

I have set my log level to `Debug` and the output is below. It appears that a sensor with a `null` value causes the entire metric reporting process to fail.

I have tried adding sensors selectively to `OhmGraphite.exe.config` under `<appSettings>` section with the `/hidden` flag to attempt to weed out the problematic values. Example of how I'm hiding them: `<add key="/nic/{2A2F495E-FA5C-4393-8549-C471BE4AD0F7}/data/2/hidden" />`

After hiding the sensors that I thought looked problematic, same issue. Would it be a better option to just ignore null sensors values (and not send them), and allow the rest of the metrics to send or am I doing something wrong here?

```
2020-09-25 12:16:23.0581|DEBUG|OhmGraphite.Program|Starting: parse config
2020-09-25 12:16:23.0919|DEBUG|OhmGraphite.Program|Finished: parse config
2020-09-25 12:16:23.1147|INFO|OhmGraphite.Program|Influxdb address: http://192.168.2.17:8086/ db: win10ohw
2020-09-25 12:16:23.1509|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-09-25 12:16:23.1509|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-09-25 12:16:23.1751|DEBUG|Topshelf.Builders.RunBuilder|Running as a console application, creating the console host.
2020-09-25 12:16:23.1751|DEBUG|Topshelf.Hosts.ConsoleRunHost|Starting up as a console application
2020-09-25 12:16:23.1751|DEBUG|OhmGraphite.MetricTimer|Starting: starting metric timer
2020-09-25 12:16:23.3035|DEBUG|OhmGraphite.SensorCollector|Hardware added: /motherboard
2020-09-25 12:16:23.3035|DEBUG|OhmGraphite.SensorCollector|Hardware added: /lpc/it8688e
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it8688e/control/0 "Fan Control #1"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it8688e/control/1 "Fan Control #2"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it8688e/control/2 "Fan Control #3"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it8688e/control/3 "Fan Control #4"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it8688e/control/4 "Fan Control #5"
2020-09-25 12:16:23.3035|DEBUG|OhmGraphite.SensorCollector|Hardware added: /lpc/it879xe
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/control/0 "Fan Control #1"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/control/1 "Fan Control #2"
2020-09-25 12:16:23.3035|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/control/2 "Fan Control #3"
2020-09-25 12:16:23.4732|DEBUG|OhmGraphite.SensorCollector|Hardware added: /intelcpu/0
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/1 "CPU Core #1"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #2"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #3"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #4"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/5 "CPU Core #5"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/6 "CPU Core #6"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/7 "CPU Core #7"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/8 "CPU Core #8"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/0 "CPU Total"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Core #3"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #4"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/4 "CPU Core #5"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/5 "CPU Core #6"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/6 "CPU Core #7"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/7 "CPU Core #8"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/8 "CPU Package"
2020-09-25 12:16:23.4732|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/9 "CPU Core #1 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/10 "CPU Core #2 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/11 "CPU Core #3 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/12 "CPU Core #4 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/13 "CPU Core #5 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/14 "CPU Core #6 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/15 "CPU Core #7 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/16 "CPU Core #8 Distance to TjMax"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/17 "Core Max"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/18 "Core Average"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/1 "CPU Core #1"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/2 "CPU Core #2"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/3 "CPU Core #3"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/4 "CPU Core #4"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/5 "CPU Core #5"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/6 "CPU Core #6"
2020-09-25 12:16:23.4837|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/7 "CPU Core #7"
[... 135 lines cut ...]
2020-09-25 12:16:29.0929|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/temperature/0 "Temperature #1"
2020-09-25 12:16:29.0929|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/temperature/2 "Temperature #3"
2020-09-25 12:16:29.0929|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/fan/0 "Fan #1"
2020-09-25 12:16:29.0929|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/fan/1 "Fan #2"
2020-09-25 12:16:29.0929|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/it879xe/fan/2 "Fan #3"
2020-09-25 12:16:29.4752|DEBUG|OhmGraphite.SensorCollector|/lpc/it8688e/control/2 did not have a value
2020-09-25 12:16:29.4752|DEBUG|OhmGraphite.SensorCollector|/lpc/it8688e/control/0 did not have a value
2020-09-25 12:16:29.4752|DEBUG|OhmGraphite.SensorCollector|/lpc/it8688e/control/1 did not have a value
2020-09-25 12:16:29.4752|DEBUG|OhmGraphite.SensorCollector|/lpc/it8688e/control/3 did not have a value
2020-09-25 12:16:29.4752|DEBUG|OhmGraphite.SensorCollector|/lpc/it8688e/control/4 did not have a value
2020-09-25 12:16:29.5137|ERROR|OhmGraphite.MetricTimer|Unable to send metrics!System.ArgumentNullException: Value cannot be null.
Parameter name: stringToEscape
   at System.Uri.EscapeDataString(String stringToEscape)
[... cut ...]
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at OhmGraphite.MetricTimer.<ReportMetrics>d__6.MoveNext()
2020-09-25 12:16:33.0044|INFO|Topshelf.Hosts.ConsoleRunHost|Control+C detected, attempting to stop service.
2020-09-25 12:16:33.0044|DEBUG|OhmGraphite.MetricTimer|Starting: stopping metric timer
```

nwilging, 2020-09-25:
wow I had looked for a similar issue for probably 15 minutes before posting this and sure enough, as soon as I opened this issue I found the similar: https://github.com/nickbabcock/OhmGraphite/issues/145

Apologies for the duplicate.

Specifying an empty `influx_password` in the config is required even if the user doesn't have a password.

closed 2020-09-25 by nwilging

nickbabcock, 2020-09-28:
No worries, I appreciate the post as it let me know that it's an easy trap for new users and that OhmGraphite should assume a blank password (ie: passwordless) user if a user is specified but a password is omitted. I'm tracking it here: #163