nickbabcock/OhmGraphite #166: Cant login to influxDB

transguy4l80, 2020-10-03:
I have tried to login to Influxdb, I know that my login info is correct based on using the same info to login to telegraf. I am trying to log in from a windows server to a linux VM on a different machine both on the same vlan of the network in my homelab. Windows firewall and linux firewall are both disabled completely but the traffic is behind a pfsense router. Any ideas would be great. 

[OhmGraphite Logs.zip](https://github.com/nickbabcock/OhmGraphite/files/5322367/OhmGraphite.Logs.zip)

transguy4l80, 2020-10-03:
Still not sure what was causing this issue however I was able to get it working by enabling a passwordless user and blank password

nickbabcock, 2020-10-04:
Thank you for providing logs. What version of influxdb are you trying to write to?

transguy4l80, 2020-10-04:
1.8

nickbabcock, 2020-10-05:
Doing some sleuthing.

In the uploaded config I noticed the following is commented out:

```xml
<!--
    <add key="influx_user" value="admin" />
    <add key="influx_password" value="*********" />
    <add key="interval" value="15" />
-->
```

You're not accidentally commenting this out, right?

Also other influxdb users have similar errors:

 - https://github.com/influxdata/influxdb/issues/8967
 - https://github.com/influxdata/influxdb/issues/16678

nickbabcock, 2020-10-24:
I'm going to assume this is resolved. Let me know if not.

closed 2020-10-24 by nickbabcock