nickbabcock/OhmGraphite #167: Update Sample Dashboards with Grafana 7 Usage

nickbabcock, 2020-10-10:
Grafana 7 has some nice visualizations like tables with inlined gauges

[image]

side note: I should be hiding those faulty sensors via https://github.com/nickbabcock/OhmGraphite#hiding-sensors

This table provides a nice, at a glance, summary of temperatures across all components. Tables like this could lead a row and be followed by the detailed breakdown of each hardware instance by time (like how it currently is).

It looks like the samples dashboards have been downloaded ~200 so some people are finding them useful. Probably a good idea to keep updated with grafana.

If anyone has panels they'd like to share, feel free to showcase them. I'll play around some more -- see what sticks. Grafana 7 is still newish, so I don't imagine I'll push this change too quickly.

nickbabcock, 2021-10-10:
Nicely done, do you want to upload the JSON here so that I can use it as a starting point?

zaibakker, 2021-10-10:
I optimized and corrected the dashboard yestherday, i was a good idea to wait for upload ;)
Source Prometheus for me, and filter with hostname,
Have fun!

[Windows Desktop Overview-1633851822152.zip](https://github.com/nickbabcock/OhmGraphite/files/7317065/Windows.Desktop.Overview-1633851822152.zip)