nickbabcock/OhmGraphite #171: NaN values causing the endpoint to fail

imbarto, 2020-11-24:
Hi, I have noticed some strange behavior on my PC, I can see in the log OG logfile that some of the sensors have an NaN value and the endpoint goes [down](https://ibb.co/52cCxV8) and is not available, prometheus can not scrape the endpoint and it is shown in grafana as a gap. 

```
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/ssd/1/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/ssd/1/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/ssd/1/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-24 14:59:24.4258|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
```
nay Idea what could be wrong ?

nickbabcock, 2020-11-25:
The link appears down. You can inline images in github if it makes it easier.

NaN values are expected to cause gaps for the metrics that reported NaN.

> and the endpoint goes down and is not available

Are you saying that OhmGraphite crashes?

imbarto, 2020-11-25:
[image]
The OhmGraphite service is up but the endpoint is down and I get gaps in all metrics like fan-speed which does not show an NaN value. 
[image]

nickbabcock, 2020-11-25:
Hmm interesting, what happens when you increase the prometheus scrape_timeout? Perhaps there is a slowdown at that time. I'll have to add some additional logging (perhaps expose internal metrics) so that a potential slowdown is more apparent.

imbarto, 2020-11-25:
I have set the scrape interval to 5s and timeout to 4s, I have also tried more variation, 10s scrape and 9s timeout, 15-14 and 20-19, I don't see this issue when I se the interval to  25s and timeout 24s and above.

nickbabcock, 2020-11-27:
Can you post the log lines from when log prints:

```
Starting: prometheus update metrics
Finished: prometheus update metrics
```

I'm interested in difference between the timestamps from these two log lines when the gap happens every 15 minutes.

imbarto, 2020-11-27:
HI, it looks like this, there are no other messages but these
`2020-11-27 00:01:16.9535|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:17.1410|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:21.9536|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:22.1270|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:26.9540|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:27.1258|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:31.9540|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:32.1805|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:36.9541|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:37.1728|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:41.9558|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:42.1033|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:46.9577|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:47.1609|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:51.9546|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:52.1904|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:01:56.9564|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:01:57.1283|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:01.9574|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:02.1307|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:06.9550|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:07.1738|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:11.9607|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:12.1340|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:16.9554|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:21.9557|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:26.9556|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:31.9557|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:36.9559|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:40.7955|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/0/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/4/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/0/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/0/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/3/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/0/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/0/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.SensorCollector|/hdd/2/load/32 had a NaN value
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:40.8111|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:02:41.9558|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:02:42.1761|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
[... cut ...]
2020-11-27 00:03:21.9574|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:03:22.1776|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:03:26.9574|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2020-11-27 00:03:27.1449|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2020-11-27 00:03:31.9576|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics`

nickbabcock, 2020-11-29:
I may need to think a bit on what the problem could be and what workarounds are available. For the problem, it could be that your [WMI](https://en.wikipedia.org/wiki/Windows_Management_Instrumentation) is slow to [return results on disk performance](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/5bd108167ddad48d824cc0c858f46772492d32de/LibreHardwareMonitorLib/Hardware/Storage/AbstractStorage.cs#L124). Hmm 🤔

imbarto, 2020-12-01:
any Idea how to debug why the endpoint is down during that time ? it does not write anything in the log. :(

nickbabcock, 2020-12-02:
You can try running the following at a powershell prompt while the endpoint has stalled:

```powershell
Get-WmiObject -Query "SELECT * FROM Win32_PerfRawData_PerfDisk_PhysicalDisk"
```

If that command takes awhile then we've narrowed down the culprit

imbarto, 2020-12-03:
Hi, 
I have run the query every second for about one hour, and usually the response is about 20-30 ms but at the time the endpoint dies the response is about 20s. 

```
Thursday, December 03, 2020 9:09:29 AM

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 0
Milliseconds      : 22
Ticks             : 225480
TotalDays         : 2.60972222222222E-07
TotalHours        : 6.26333333333333E-06
TotalMinutes      : 0.0003758
TotalSeconds      : 0.022548
TotalMilliseconds : 22.548

Thursday, December 03, 2020 9:09:30 AM

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 19
Milliseconds      : 440
Ticks             : 194400975
TotalDays         : 0.000225001128472222
TotalHours        : 0.00540002708333333
TotalMinutes      : 0.324001625
TotalSeconds      : 19.4400975
TotalMilliseconds : 19440.0975

Thursday, December 03, 2020 9:09:51 AM

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 0
Milliseconds      : 21
Ticks             : 219998
TotalDays         : 2.54627314814815E-07
TotalHours        : 6.11105555555555E-06
TotalMinutes      : 0.000366663333333333
TotalSeconds      : 0.0219998
TotalMilliseconds : 21.9998
```

nickbabcock, 2020-12-04:
Excellent work. Thank you for confirming the issue. I don't know what the solution is but this is quite useful!

PR #221 (2021-05-14): Update LibreHardwareMonitor with imposed WMI timeouts
Closes #171
closed 2021-05-14 by nickbabcock