nickbabcock/OhmGraphite #172: Metric names are occasionally clobbered

ikorb, 2020-11-25:
I observed occasional clobbered (truncated or mixed) metric names in my Carbon database, e.g.:

    ./ohm/DRAGON/gpu-nvidia/0/temperature/gohm/DRAGON/nic/266FD541-153A-4BCF-B510-CAD4858E01AA/throughput/uploadspeed.wsp
    ./rature/temperature/2.wsp
    ./ohm/DRAoltage/+5v.wsp
    ./ohm/DRA/DRAGON/gpu-nvidia/0/power/gpupackage.wsp
    ./ohm/DRAGO/2270CAF9-24FF-4E2F-AFD5-FBB23C11E87F/data/datauploaded.wsp

Based on the time stamps of the database files, I found that each of them corresponds to a log entry of OhmGraphite where the reported time for sending is longer than the polling interval, e.g.:

    2020-11-23 20:33:51.1512|INFO|OhmGraphite.MetricTimer|Sent 133 metrics in 5262,329ms
    2020-11-24 00:02:57.0687|INFO|OhmGraphite.MetricTimer|Sent 133 metrics in 5140,7891ms

There are no log entries other than the once-per-interval Sent message. I'm currently waiting for the problem to trigger once again with Debug log level and will update this issue if anything suspicious shows in the log.

nickbabcock, 2020-11-25:
Good find. I believe corruption is possible (at least for the graphite writer) if it takes longer to write the data than the set interval as two sensors collections are written intertwined on a single connection. I can work on fixing this issue. The solution that I can implement will ensure that only one sensor collection sees the connection at a time. The downside of this approach is that if sending the data will always be slower than the interval, there may be a unbounded growth of threads waiting to send data, but I believe this to be a small and not important use case.

ikorb, 2020-11-25:
Sounds good, although limiting the number of waiting threads is probably a good idea to avoid DoSsing the system if there is a persistent network problem.

PR #175 (2020-11-29): Fix graphite connection write contention
If polling for sensors + writing to graphite takes longer than the
configured interval, then it is possible for two threads to be writing
data to the same graphite connection. This will cause corrupted data to
be sent to graphite.

The fix is the limit the number of threads that have access to the
connection to 1. Other threads that would have caused contention are
forced to wait outside. To ensure there is not an unbounded number of
threads waiting to write, if they can't acquire a lock in a second then
they jettison the write attempt.

Closes #172
closed 2020-11-29 by nickbabcock