nickbabcock/OhmGraphite #173: Change of graphicscard

hnktnk, 2020-11-26:
I have bought a new GPU, moved from a Geforce 660 -> GTX 1070 and now my graphs doesn't get updated. Do I need to change anything in the configuration os something else?

nickbabcock, 2020-11-26:
Moving within nvidia -- nothing should be changed. What do the logs say on startup on what sensors are detected?

There will be lines like:

```
OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/3 "GPU Memory"
```

hnktnk, 2020-11-26:
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/temperature/0 "GPU Core"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/0 "GPU Core"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/1 "GPU Memory"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/clock/2 "GPU Shader"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/fan/0 "GPU"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/0 "GPU Core"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/1 "GPU Memory Controller"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/2 "GPU Video Engine"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/4 "GPU Bus"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/control/0 "GPU Fan"
2020-11-26 21:53:53.8557|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/smalldata/3 "GPU Memory Total"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/smalldata/2 "GPU Memory Used"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/smalldata/1 "GPU Memory Free"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/load/3 "GPU Memory"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/power/0 "GPU Package"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/throughput/0 "GPU PCIe Rx"
2020-11-26 21:53:53.8657|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-nvidia/0/throughput/1 "GPU PCIe Tx"

Plus, I stopped the service and started it manually to see the log-output and voila, the graphs are now populated again.

You should have asked "have you tried turning it on and off" :)

nickbabcock, 2020-11-26:
Ah yes, restarts -- the best solution 😆

closed 2020-11-26 by nickbabcock