nickbabcock/OhmGraphite #182: InfluxDB v2 Support

fndmossmann, 2020-12-10:
Hi,

Now that InfluxDB v2 is out for some time, there are any plans supporting the new version?

nickbabcock, 2020-12-10:
Yes support for v2 is planned. Looks like v2 has been out for a month so I'm not in a rush to support it. Maybe they'll introduce a backwards compatible client? Not sure what's changed or why the client needed a breaking change.

fndmossmann, 2020-12-10:
Nice!
I have both versions running in parallel, for testing purposes.
OhmGraphite its the only one that can't write directly to InfluxDB v2.

As I was writing this response, I had an idea that pretty much solves this issue for the time being.
Which is use Telegraf as proxy.
My telegraf conf already outputs to both DBs, so I just added a listener:

```
 [[inputs.influxdb_listener]]
  service_address = ":8186"
  read_timeout = "10s"
  write_timeout = "10s"
  max_body_size = 0
  basic_username = "foobar"
  basic_password = "barfoo"
```

And changed the Ohm config so it can write to Telegraf and not directly do InfluxDB:
```
  <appSettings>
    <add key="type" value="influxdb" />
    <add key="interval" value="5" />
    <add key="influx_address" value="http://127.0.0.1:8186" />
    <add key="influx_db" value="foo_db" />
    <add key="influx_user" value="foobar" />
    <add key="influx_password" value="foobar" />
    <add key="interval" value="5" />
  </appSettings>
```
And its working nicely!
I see data in both DBs, as expected.

mkr4, 2021-01-21:
When I try to collect metrics with telegraf from prometheus page it warns me with deprecation message too.
> W! [inputs.prometheus] Use of deprecated configuration: 'metric_version = 1'; please update to 'metric_version = 2'

Telegraf config:
```toml
[[inputs.prometheus]]
  urls = ["http://127.0.0.1:4445/metrics"]
  namedrop = ["ohm_*"]
```

TomAvecLeVdsl, 2021-02-18:
Hi, replying to this thread trying to setup the telegraf proxy, What database name do i have to specify to make it work ?
Thanks for taking the time to reply

gon3456, 2021-04-01:
I'm using InfluxDB v2, so I'm really looking forward to this support.

nickbabcock, 2021-04-03:
So I spent a few hours on this problem. Influxdb 2.0 docs [claim they have 1.x write compatibility](https://docs.influxdata.com/influxdb/cloud/reference/api/influxdb-1x/write/), but I am unable to replicate this. I have a [PR](#205) that follows the docs, but I am still receiving unauthorized responses. Frustrating to be sure. I will need to reach out to the influx community as I'm scratching my head as to how to continue.

nickbabcock, 2021-05-11:
If you'd like to test out and provide feedback on the influxdb 2 support before it is officially released, checkout the latest artifact: https://github.com/nickbabcock/OhmGraphite/suites/2706011826/artifacts/59649256

Sample config:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="influx2" />
    <add key="influx2_address" value="http://localhost:8086" />
    <add key="influx2_org" value="myorg" />
    <add key="influx2_bucket" value="mydb" />
    <add key="influx2_token" value="thisistheinfluxdbtoken" />
    <add key="interval" value="5" />
  </appSettings>
</configuration>
```