nickbabcock/OhmGraphite #185: No power for nvidia gpu (it has worked for a while)

ikorbln, 2020-12-17:
Hello, i think after a couple driver updates from nvidia i did not get any power readings anymore for the gpu.
I the logfile there are all sorts of readings for load/clock but no power.

I use the latest version 0.16.
Nvidia is 460.89 / RTX3070

nickbabcock, 2020-12-17:
Can you check if `%windir%/system32/nvml.dll` exists

Is the power readings present if you run the [latest version of LibreHardwareMonitor (at the time of writing)](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/suites/1634426889/artifacts/30185229)?

ikorbln, 2020-12-17:
The dll is not in this directory and with the Libre there is no power reading for gpu.

nickbabcock, 2020-12-17:
Hmm I would recommend reinstalling nvidia geforce or compiling nvml via some other means and placing it in that directory or alongside the ohmgraphite executable.

ikorbln, 2020-12-17:
ok, after reinstalling the driver the nvml.dll is now in the folder but the Libre did not show the power.

nickbabcock, 2020-12-17:
I'd next try restarting and then copying the nvml.dll to the same folder as Libre / OhmGraphite

ikorbln, 2020-12-17:
sorry, no success. Last day i saw the gpu power in the ohmgraphite_log was the 8 Dec.

nickbabcock, 2020-12-18:
I wonder what happened on December 8th.

peterholmberg, 2020-12-28:
New user of your exporter and I'm also experiencing this. 

Running v.0.16 of Ohm and v.0.8.6 of libre. The power usage is not reported in any of them. Might be a change to the nvidia api?

nickbabcock, 2020-12-28:
Good find. Looks like we may have to wait for nvidia to push this fix or have users download their hotfix

nickbabcock, 2021-01-27:
Fixed in the latest nvidia driver update. After updating, restart ohmgraphite.

closed 2021-01-27 by nickbabcock