nickbabcock/OhmGraphite #186: No flag to specify config location

andreykaipov, 2020-12-27:
Hi - great tool. I love it! :)

I tried running `--help` to see the available options but couldn't find a flag for specifying the location of the config file when running the binary. For example:

```ps1
PS D:\> .\Monitor\programs\OhmGraphite-0.16.0\OhmGraphite.exe --config .\Far\Far\Away\config.xml
```

I'm not familiar with Topshelf or .NET development in general, so hopefully adding functionality for custom flags isn't too involved.

nickbabcock, 2020-12-27:
It's a bit more involved than it should be, iirc, as OhmGraphite can be ran interactively or installed as a service. And it is not clear to me how one would allow custom options at install time propagate to when the service is ran. But maybe there is an easier solution -- what is your use case?

andreykaipov, 2020-12-28:
Oh I'd just like to keep the configuration outside of the location I installed the binary. For example, I'd like to move OhmGraphite over to `C:\Program Files`, but keep my system's configuration for it in source control in a different directory. 

I'm actually not installing it as a service but instead starting it up "interactively" via the Task Scheduler under the SYSTEM user during system startup, so it's not clear to me either what the functionality for custom options should be when installing OhmGraphite as a service.

Using the Task Scheduler is probably not the best approach, but I like it for consistency as I can manage OhmGraphite, Prometheus, and Grafana all through the Task Scheduler instead of installing NSSM for both Prometheus and Grafana as that's what they recommend for running their tools as Windows services.

nickbabcock, 2020-12-29:
Hmm yes, let me think about this. My first thought is that your use case can be supported. I'll play around with this.

closed 2021-01-04 by nickbabcock

andreykaipov, 2021-04-03:
Sorry for the super late response but I really appreciate you adding this feature! Thank you!

nickbabcock, 2021-04-03:
You're welcome, sorry it took me so long to release it!

PR #187 (2021-01-04): Allow specifying config path when ran interactive
When running interactively, OhmGraphite can be passed a config path like
so:

```
.\OhmGraphite.exe run -config:"C:\apps\OhmGraphite.exe.config"
```

Closes #186