nickbabcock/OhmGraphite #188: grafana 7.4.0+ requires changes in template variables (prometheus)

swiffer, 2021-01-09:
grafana 7.4.0+ requires using `?<value>` in regex for variables:

e.g.

/ohm_(?<value>\w+)_volts.*/

[image]

nickbabcock, 2021-01-09:
Thank you for the report. Is there any context for this change (like a PR or changelog entry), so I can familiarize myself as it seems odd to introduce (what seems like) a breaking change.

Do you know of a way that we can structure dashboard variables so they don't break on 7.4.0+?

swiffer, 2021-01-09:
https://grafana.com/docs/grafana/latest/variables/filter-variables-with-regex/#filter-and-modify-using-named-text-and-value-capture-groups

But I guess it's more a regression in grafana. Reading the paragraphs above the way you are using it currently should still work just fine, right?

nickbabcock, 2021-01-20:
Saw that the first 7.4.0 beta has been released. I'm curious if this this issue has been fixed. Have you tested this, @swiffer ?

swiffer, 2021-01-25:
yes, it still happens in 7.4.0-beta1 (reimported the dashboard from your repo)

[image]

nickbabcock, 2021-01-26:
Reported upstream: https://github.com/grafana/grafana/issues/30617

nickbabcock, 2021-01-27:
Confirmed grafana issue and looks like it will be fixed before the release, so I'm closing the issue.

closed 2021-01-27 by nickbabcock