nickbabcock/OhmGraphite #189: Clickhouse database

denser, 2021-01-12:
Hello, is anyone wanted to use clickhouse to store data?

I am new with C++ and any advice how to adopt current writes for CH?
I suppose it must be very close to InfuxDB, no?

nickbabcock, 2021-01-12:
I'm not completely familiar with clickhouse. Which output interface should be used? I'm assuming Mysql or HTTP but I'm not sure

Normally I try and reference other metric exporters such as collectd, telegraf, or another metric library, but none of them seem to have clickhouse as a supported sink 🤔

denser, 2021-03-03:
I am already find build of graphite wit clickhouse base, thank you for an answer!

https://github.com/lomik/graphite-clickhouse

Database make so tiny now. :)

closed 2021-03-03 by denser