nickbabcock/OhmGraphite #191: Adding to grafana

shandshellin, 2021-01-13:
no errors in ohmgraphite and i can view the metrics in a web browser on my network. What i cannot seem to do is get the data into grafana. Is there a special way to add a data source for it or some other config i'm not seeing. Using Prometheus.

nickbabcock, 2021-01-13:
Seems like you're looking for prometheus + grafana tutorials as this doesn't appear to be ohmgraphite specific:

- https://prometheus.io/docs/visualization/grafana/
- https://grafana.com/go/webinar/getting-started-with-grafana/

I've also updated the readme to be more explicit about how to configure prometheus

Thanks

closed 2021-01-13 by nickbabcock

shandshellin, 2021-01-14:
What http address would i use in prometheus to add this? Would i use Prometheus as the data port or add a graphite one with the prometheus port configured?