nickbabcock/OhmGraphite #193: Netdata prometheus import issue

mkr4, 2021-01-19:
Hello.
I have an issue with netdata's prometheus importer.

Looks like it can't parse spaces in sensor id.

Could you add an `alias` feature for hardware like you did to sensor's name?

Error log:

```
[ WARN  ] prometheus[win_ohm] collect_gauge_counter.go:41 error on adding dim to chart 'ohm_cpu_celsius' : unacceptable symbol in dim ID 'ohm_cpu_celsius|hardware=Intel Core i7-4790K,hw_instance=0,sensor=Core Average' : ' '
[ WARN  ] prometheus[win_ohm] collect_gauge_counter.go:41 error on adding dim to chart 'ohm_cpu_celsius' : unacceptable symbol in dim ID 'ohm_cpu_celsius|hardware=Intel Core i7-4790K,hw_instance=0,sensor=CPUCoreTemp1' : ' '
[ WARN  ] prometheus[win_ohm] collect_gauge_counter.go:41 error on adding dim to chart 'ohm_cpu_celsius' : unacceptable symbol in dim ID 'ohm_cpu_celsius|hardware=Intel Core i7-4790K,hw_instance=0,sensor=CPU Core #4 Distance to TjMax' : ' '
[ WARN  ] prometheus[pen7i_ohm] collect_gauge_counter.go:41 error on adding dim to chart 'ohm_cpu_celsius' : unacceptable symbol in dim ID 'ohm_cpu_celsius|hardware=Intel Core i7-4790K,hw_instance=0,sensor=CPU Core #3' : ' '
[ WARN  ] prometheus[pen7i_ohm] collect_gauge_counter.go:41 error on adding dim to chart 'ohm_cpu_celsius' : unacceptable symbol in dim ID 'ohm_cpu_celsius|hardware=Intel Core i7-4790K,hw_instance=0,sensor=CPU Core #2' : ' '

```

CPUCoreTemp1 has been taken from config: `<add key="/intelcpu/0/temperature/0/name" value="CPUCoreTemp1" />`

Ohm log:
```
2021-01-19 18:17:38.8940|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2021-01-19 18:17:38.8940|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2021-01-19 18:17:38.8960|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Core #3"
2021-01-19 18:17:38.8960|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #4"
```

Exported metrics:
```
ohm_cpu_celsius{hardware="Intel Core i7-4790K",sensor="CPU Package",hw_instance="0"} 50
ohm_cpu_celsius{hardware="Intel Core i7-4790K",sensor="CPU Core #2",hw_instance="0"} 50
ohm_cpu_celsius{hardware="Intel Core i7-4790K",sensor="CPUCoreTemp1",hw_instance="0"} 48
ohm_cpu_celsius{hardware="Intel Core i7-4790K",sensor="CPU Core #4",hw_instance="0"} 46
ohm_cpu_celsius{hardware="Intel Core i7-4790K",sensor="CPU Core #3",hw_instance="0"} 48
```

nickbabcock, 2021-01-20:
I'm open to this idea. I'm not familiar with netdata. Not able to display spaces seems like a bit of a bug on netdata's side, no?

mkr4, 2021-01-21:
I don't know if it's a bug or a feature. But it was in there since the [initial commit](https://github.com/netdata/go-orchestrator/blob/4037058e53446ce8af61a4821f3427d12c842327/module/charts.go#L395):
```go
	if id := checkID(chart.ID); id != -1 {
		return fmt.Errorf("unacceptable symbol in ID : '%c'", id)
	}

func checkID(id string) int {
	for _, r := range id {
		if unicode.IsSpace(r) {
			return int(r)
		}
	}
	return -1
}
```
(from [go.d.plugin](https://github.com/netdata/go.d.plugin/blob/bd69d2d40303f7c937086900d1b94466ae1efd6e/agent/module/charts.go#L433))