nickbabcock/OhmGraphite #206: InfluxDB - Allow Invalid SSL Cert

ehbush, 2021-04-23:
Is there a way to allow invalid SSL certs on the InfluxDB side? Didn't see it in the documentation, nor in the code from InfluxConfig file. Hoping its something that can be added, as I am sooooo close to being able to use this :-D 

Great work, regardless, love it!

nickbabcock, 2021-04-24:
The option would need to be added, I'd like to mull over the best way to integrate this ask. I know in past projects I used a three state field to represent the possibility of:

- Validate against system certificates (the default)
- Do not validate certificate (accept all)
- Validate against the certificate pointed to by a file path

Ideally whatever is introduced would be sufficient for all use cases (ie: one certificate format in PEM encoding is sufficient to configure all sinks), but this is something that need to be looked into. I'd imagine that the server use case (prometheus) would require a different set of options due to needing the private key as well.

ehbush, 2021-04-26:
For sure. This is specific to InfluxDB for me, and InfluxDB 1.X. For example, in telegraf's config file: 

```
 ## Optional TLS Config for use on HTTP connections.
  # tls_ca = "/etc/telegraf/ca.pem"
  # tls_cert = "/etc/telegraf/cert.pem"
  # tls_key = "/etc/telegraf/key.pem"
  ## Use TLS but skip chain & host verification
   insecure_skip_verify = true
```

ehbush, 2021-05-09:
Any additional thoughts @nickbabcock?

nickbabcock, 2021-05-09:
Yeah I'm planning on looking into this more once #214 and #182 is solved. Influxdb is prone to breaking changes so I don't want to make the solution influxdb specific.

PR #218 (2021-05-11): Configurable certificate verification
This commit introduces the `certificate_verification` property:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="certificate_verification"
         value="C:\apps\OhmGraphite\influxdb-selfsigned.crt" />
   </appSettings>
</configuration>
```

The possible values:

 - True (the default): all certificates are verified
 - False: No certificates are verified (INSECURE)
 - a file path of a certificate that the server is allowed to return and
 still be considered a valid request (useful for self signed
 certificates). [... cut]