nickbabcock/OhmGraphite #214: 0.18.0 not working

swiffer, 2021-05-08:
0.18.0 is no longer starting correctly, see log below

```
2021-05-08 20:53:57.2373|ERROR|OhmGraphite.PrometheusServer|Exception: starting prometheus server!System.EntryPointNotFoundException: Entry point was not found.
   at LibreHardwareMonitor.Interop.Kernel32.CopyMemory(IntPtr Destination, IntPtr Source, UInt32 Length)
   at LibreHardwareMonitor.Hardware.Storage.NVMeIntel.IdentifyController(SafeHandle hDevice, NVME_IDENTIFY_CONTROLLER_DATA& data)
   at LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(StorageInfo storageInfo)
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(StorageInfo storageInfo, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(String deviceId, UInt32 driveNumber, UInt64 diskSize, Int32 scsiPort, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at OhmGraphite.SensorCollector.Start()
   at OhmGraphite.PrometheusServer.<Start>b__4_0()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
2021-05-08 20:53:57.2373|ERROR|OhmGraphite.Program|OhmGraphite TopShelf encountered an error!System.Exception: Exception with starting prometheus server
 ---> System.EntryPointNotFoundException: Entry point was not found.
   at LibreHardwareMonitor.Interop.Kernel32.CopyMemory(IntPtr Destination, IntPtr Source, UInt32 Length)
   at LibreHardwareMonitor.Hardware.Storage.NVMeIntel.IdentifyController(SafeHandle hDevice, NVME_IDENTIFY_CONTROLLER_DATA& data)
   at LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(StorageInfo storageInfo)
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(StorageInfo storageInfo, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(String deviceId, UInt32 driveNumber, UInt64 diskSize, Int32 scsiPort, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at OhmGraphite.SensorCollector.Start()
   at OhmGraphite.PrometheusServer.<Start>b__4_0()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- End of inner exception stack trace ---
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   at OhmGraphite.PrometheusServer.Start()
   at OhmGraphite.Program.<>c.<Main>b__1_5(IManage tc)
   at Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   at Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   at Topshelf.Runtime.Windows.WindowsServiceHost.OnStart(String[] args)
2021-05-08 20:53:57.2373|FATAL|Topshelf.Runtime.Windows.WindowsServiceHost|The service did not start successfully!System.Exception: Exception with starting prometheus server
 ---> System.EntryPointNotFoundException: Entry point was not found.
   at LibreHardwareMonitor.Interop.Kernel32.CopyMemory(IntPtr Destination, IntPtr Source, UInt32 Length)
   at LibreHardwareMonitor.Hardware.Storage.NVMeIntel.IdentifyController(SafeHandle hDevice, NVME_IDENTIFY_CONTROLLER_DATA& data)
   at LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(StorageInfo storageInfo)
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(StorageInfo storageInfo, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(String deviceId, UInt32 driveNumber, UInt64 diskSize, Int32 scsiPort, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at OhmGraphite.SensorCollector.Start()
   at OhmGraphite.PrometheusServer.<Start>b__4_0()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- End of inner exception stack trace ---
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   at OhmGraphite.PrometheusServer.Start()
   at OhmGraphite.Program.<>c.<Main>b__1_5(IManage tc)
   at Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   at Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   at Topshelf.Runtime.Windows.WindowsServiceHost.OnStart(String[] args)
2021-05-08 20:53:57.2453|ERROR|Topshelf.HostFactory|The service terminated abnormally!System.Exception: Exception with starting prometheus server
 ---> System.EntryPointNotFoundException: Entry point was not found.
   at LibreHardwareMonitor.Interop.Kernel32.CopyMemory(IntPtr Destination, IntPtr Source, UInt32 Length)
   at LibreHardwareMonitor.Hardware.Storage.NVMeIntel.IdentifyController(SafeHandle hDevice, NVME_IDENTIFY_CONTROLLER_DATA& data)
   at LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(StorageInfo storageInfo)
   at LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(StorageInfo storageInfo, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(String deviceId, UInt32 driveNumber, UInt64 diskSize, Int32 scsiPort, ISettings settings)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at OhmGraphite.SensorCollector.Start()
   at OhmGraphite.PrometheusServer.<Start>b__4_0()
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   --- End of inner exception stack trace ---
   at OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
   at OhmGraphite.PrometheusServer.Start()
   at OhmGraphite.Program.<>c.<Main>b__1_5(IManage tc)
   at Topshelf.ServiceConfiguratorExtensions.<>c__DisplayClass2_0`1.<WhenStarted>b__0(T service, HostControl control)
   at Topshelf.Builders.DelegateServiceBuilder`1.DelegateServiceHandle.Start(HostControl hostControl)
   at Topshelf.Runtime.Windows.WindowsServiceHost.OnStart(String[] args)
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(Object state)
--- End of stack trace from previous location ---
   at System.ServiceProcess.ServiceBase.Run(ServiceBase[] services)
   at System.ServiceProcess.ServiceBase.Run(ServiceBase service)
   at Topshelf.Runtime.Windows.WindowsServiceHost.Run()
   at Topshelf.HostFactory.Run(Action`1 configureCallback)
```

nickbabcock, 2021-05-08:
Bummer, I feared something like this would happen. And funny enough my main setup is prometheus, so I'm not sure what is the difference between the two. I'm not quite sure how to debug something like this as the new executable should be self contained. Hmm, I'll mull this over and see if there is a quick fix else I'll roll back to the old method. Thank you for reporting so promptly.

swiffer, 2021-05-08:
Mh, I'm using Windows 10 64bit 21H1 Release Preview at the moment, and it's always the Intel NVME part where it is throwing the error...

nickbabcock, 2021-05-08:
Can you try this: https://github.com/nickbabcock/OhmGraphite/suites/2687684966/artifacts/59247931 . It was produced from #215 . Heads up,  it is a zip in a zip annoyingly. Perhaps some key component is being trimmed out of the final executable

nickbabcock, 2021-05-09:
I think I fixed the issue, can you try https://github.com/nickbabcock/OhmGraphite/suites/2691329201/artifacts/59313442 (double zip again).

swiffer, 2021-05-09:
@nickbabcock - working fine here!

PR #216 (2021-05-09): Update LibreHardwareMonitor to fix CopyMemory
This fixes the systems that use the intel nvme drives. Closes #214
closed 2021-05-09 by nickbabcock