nickbabcock/OhmGraphite #223: Asus Embedded Controller (Asus EC) sensor not deteced

bigfist, 2021-05-27:
Hello,

first of all, thx for OhmGraphite. It is amazing and i really like it in grafana (with influxDB)
To my problem: 
I have an asus x570 rog gaming e motherboard. On it, there is a pin for temperature sensor. There  i have connected a temp senosr from my waterpump.
Unfortunately OhmGraphite does not show it at all.
In HWInfo it is listed as following:
[image]

Does anybody know a way to add the sensor to OhmGraphite and grafana?

nickbabcock, 2021-05-28:
It may not yet be supported. The good news is that this is all open sourced and OhmGraphite exposes all sensors detected by LibreHardwareMonitor so you should be able to search [if there is already an open issue](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues). I think I found an open issue: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/282

bigfist, 2021-05-28:
thanks for reply. I will try out if i can write some code to add drivers to ohmgraphite

nickbabcock, 2024-01-29:
Closing this out as potentially already supported, please let me know if otherwise.

closed 2024-01-29 by nickbabcock