nickbabcock/OhmGraphite #227: Question on network port utilization and the ability to enable only specific monitoring.

shatai1, 2021-06-22:
Hi there,

First off, this is an amazing tool, thank you for all your work on this :)

I've noticed that the process ohmgrahite has multiple open network connections as show in screenshot, is this normal please?
[image]

Is it possible to enable only specific sensor, say only CPU temperature? If yes how do I do that please?

Thank you very much for your kind support.

nickbabcock, 2021-06-22:
Does the number of open connections grow over time, or is it always approximately 10? I'm not too concerned unless you find it poses an issue in which case we can find a solution.

[See how to hide sensors](https://github.com/nickbabcock/OhmGraphite#hiding-sensors) if you want to disable them.

shatai1, 2021-06-22:
Hi there! Thank you so much for the quick reply :) I have a few more questions please.

Does Ohmgraphite support HTTPS? I have https enabled on my influxDB and getting the follow error in Ohmgraphite logs
[image]

Is the following the correct way to hide sensors?
[image]

This is what's showing in my logs even after adding hidden suffix.
[image]

nickbabcock, 2021-06-23:
> Does Ohmgraphite support HTTPS?

Yes: https://github.com/nickbabcock/OhmGraphite#certificates

> Is the following the correct way to hide sensors?

Yes that looks correct, the log lines that you see represent more that the sensor was detected, but your hidden line should stop the those sensors from reporting data into influxdb.

shatai1, 2021-06-23:
Thanks for the confirmation!

Do you have any thoughts on why I may be getting the following error please?

[image]

I have tried all of the following, still no luck, the certificate is self-signed.

[image]

[image]

[image]

nickbabcock, 2021-06-24:
Sorry, I am unable to reproduce. Feel free to include reproducible steps (eg: cert generation + docker commands)

shatai1, 2021-06-24:
No worries and thank you so much for your kind help support so far :)  I'll have HTTPS disabled for now.

One more question in regards to hiding sensor please.  At the moment the following sensors are discovered on my laptop.

[image]

As I am planning to deploy OhmGraphite on several machines, each machine would have different NIC id, is it possible to use something equivalent to wildcard "*" to hide the NIC as against to having to provide the exact NIC ID please?

[image]

nickbabcock, 2021-06-25:
That is a good idea, let me think about how that can be done.

shatai1, 2021-06-25:
Thank you Nick, really appreciate all your help! Hope to hear some good news soon :)

nickbabcock, 2021-06-25:
Can you confirm that this artifact from PR #229 implements this suggestion to your satisfaction? https://github.com/nickbabcock/OhmGraphite/suites/3084649803/artifacts/70404608

shatai1, 2021-06-28:
Hi Nick, yes! Works great now!  Thank you so much for this, you're awesome!

PR #229 (2021-07-22): Allow hiding of sensors by glob
One can use globs to ignore a group of sensors. For instance, to hide
all power sensors and hide clock sensors from an AMD CPU:

```xml
<add key="/amdcpu/*/clock/*/hidden" />
<add key="/*/power/*/hidden" />
```

Closes #227
closed 2021-07-22 by nickbabcock