nickbabcock/OhmGraphite #246: Sync remaining sample dashboards to latest version

nickbabcock, 2021-10-12:
The prometheus sample dashboard was updated in https://github.com/nickbabcock/OhmGraphite/pull/245, so the rest should be updated as well.

- [x] prometheus
- [ ] graphite
- [ ] influxdb
- [ ] postgres / timescale

swiffer, 2021-11-30:
Hi @nickbabcock - used the latest dashboard from master to see if things are fixed - drives only showing once now - sadly there still was no repeat for multiple harddrives

i searched a little bit and found info that for repeat to work you need to enable the include all option and use this as the variable dashboard default value. it is working fine for me

https://community.grafana.com/t/repeat-panel-not-repeating/17818

here is how it is configured now for me:

[image]

i also noted that some of the variables had been set to on dashboard load while others had been on time range change.

for the nic i simply resized my existing panel - thanks for you great work on these dashboards!

nickbabcock, 2021-11-30:
@swiffer perfect timing, I was just doing some additional hardware testing today! I filed a couple PRs that update the sample dashboard and fix the issues you pointed out.