nickbabcock/OhmGraphite #269: Grafana - get metrics directly from OhmGraphite

Tualua, 2021-12-01:
Hello! Is there any way to show metrics on Grafana without installing Prom/Graphite/etc?

For example I have a server with couple VMs.
I want to install Grafana to server and OhmGraphite in VMs and det dashboards for each vm.

Is it possible?

nickbabcock, 2021-12-01:
Technically yes, but realistically no. There is a JSON data source plugin for Grafana where OhmGraphite could expose the instantaneous values. However to make any sort of visualization over time, one needs to store metrics, which is where a metric store comes into play.

At this time, I don't find it worthwhile to code a new, integrated metric store. As it would be more buggy than the other production-worthy stores, and suffer from either the inflexibility of the JSON data source plugin or whatever metric interface the integrated store chose to expose (eg: PromQL, Graphite, etc).

If you really push it, one day it may be feasible to embed a SQLite database on each machine and receive commands from Grafana, but even then it's not architecturally ideal. It makes a bit more sense for grafana to always reach out to a single spot for all metrics (eg: the prometheus instance) instead of needing to connect directly to each client. If a client is off then all metric data for that client is unavailable.

As for your setup, I would recommend installing one of the currently supported metric stores. Prometheus and influxdb are quick to install and easy to maintain. You only need one instance of them running.

Tualua, 2021-12-01:
Thanks for answer. I understand now

nickbabcock, 2021-12-01:
Excellent, good to hear.

closed 2021-12-01 by nickbabcock