nickbabcock/OhmGraphite #277: PSU (Corsair RM750i) sensors appear in LHM but not in OhmGraphite

arturhgca, 2022-01-16:
Hi!

Thank you very much for your work on this tool, it's working great for my use case! However, there is one piece of hardware that isn't being detected by OhmGraphite (v0.23.1), even though it appears in LHM (v0.8.9): my Corsair RM750i PSU.

This is what LHM shows:

[image]

I've checked the OhmGraphite logs and there are no sensors matching these (but other sensors in the system are fine). This is my config:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="influx2" />
    <add key="influx2_address" value="http://localhost:8086" />
    <add key="influx2_org" value="omitted" />
    <add key="influx2_bucket" value="omitted" />
    <add key="influx2_token" value="omitted" />
    <add key="interval" value="5" />
  </appSettings>
</configuration>
```

I have also tried removing all keys from the config, but that changed nothing regarding sensors.

Any idea of what might be happening?

Thanks!

nickbabcock, 2022-01-16:
Thanks for the reporting this issue. Would you like to try out [the build](https://github.com/nickbabcock/OhmGraphite/suites/4954656669/artifacts/143934746) from #278 and see if that fixes your issue?

arturhgca, 2022-01-16:
Sure thing, on it. Thanks for the quick response!

arturhgca, 2022-01-16:
@nickbabcock the new build works! I can see the sensors being added in the logs and in InfluxDB.

nickbabcock, 2022-01-16:
Glad to hear. You can keep using that nightly build and the next release will include this functionality.

PR #278 (2022-02-18): Add PSU metric reporting
[In June of 2021, LibreHardwareMonitor gained support for reporting
metrics from the PSU][0]. This commit exposes this functionality by
default in OhmGraphite. One can opt out of sending PSU metrics with:

```xml
<add key="/controller/enabled" value="false" />
```

Closes #277 

[0]: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/458
closed 2022-02-18 by nickbabcock