nickbabcock/OhmGraphite #280: Start command get errors

Impre-visible, 2022-01-25:
I guys, i want to use OhmGraphite, but when i do the start command i have errors in the log files :
https://pastebin.com/7Xh3Yuys
The exact command is : ``.\OhmGraphite.exe start``

nickbabcock, 2022-01-25:
Thank you for the report. In it, I see:

> The configuration section cannot contain a CDATA or text element

This makes it sound like there is an issue with the configuration file. Potentially it is malformatted XML. You can check if it is valid by copying and pasting the config at xmllint.com: https://xmllint.com/

If it's still valid, you can paste it here (redact what's necessary) so we can take a look at the config.

Impre-visible, 2022-01-26:
i think the problem is because the config file look like this : 
```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
    <appSettings>
        <add key="type" value="influxdb" />
        <add key="interval" value="5" />
        <add key="influx_address" value="http://192.168.0.87:8086" /><add key="influx_db" value="PCStats" />
-->
    </appSettings>
</configuration>
```
I removed the --> i will try

Impre-visible, 2022-01-26:
that was the error everything work, thanks :D

closed 2022-01-26 by Impre-visible