nickbabcock/OhmGraphite #298: running latest but missing ohm_hdd_bytes

phynias, 2022-06-11:
loaded latest ohm and dashboard. the dash has some graphs that use ohm_hdd_bytes, but i don't see that in prometheus.

nickbabcock, 2022-06-11:
I believe there's a regression in the underlying sensor library, LibreHardwareMonitor. Do you have a samsung nvme drive (ie: a 980 PRO)? If so, I'll fix the issue upstream and make another release.

phynias, 2022-06-11:
no, mushkin reactor ssd

nickbabcock, 2022-06-12:
Bummer, if you run the [LibreHardwareMonitor GUI](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor#where-can-i-download-it) do you see values for data read / data written for you ssd? If not, then it looks like it may be an issue with properly detecting these values in LibreHardwareMonitor

phynias, 2022-06-12:
looks ok in there.
[image]

nickbabcock, 2022-06-12:
Yeah looks like your hard drive isn't exposing any data fields

[image]

This is why the `ohm_hdd_bytes` graphs are blank as they are rely on the data sensors.

phynias, 2022-06-13:
ahh ok. weird none of my drives have a data section.

closed 2022-07-30 by nickbabcock