nickbabcock/OhmGraphite #314: aquacomputer highflow next sensor

EnlightedBitFox, 2022-10-07:
Hey,
is there a way how to get the sensor data from aquacomputer highflow next sensor in OhmGraphite?

nickbabcock, 2022-10-07:
The aqua computer sensors should be enabled by default. Do you not see them if you use LibreHardwareMonitor directly? https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases

EnlightedBitFox, 2022-10-07:
Unfortunately not.
[image]

nickbabcock, 2022-10-07:
Ah bummer, well the first step would be to raise an issue in the LibreHardwareMonitor repo to add support for the highflow sensor. Then once it has been added upstream, it'll be exposed in ohmgraphite.

EnlightedBitFox, 2022-10-07:
> Ah bummer, well the first step would be to raise an issue in the LibreHardwareMonitor repo to add support for the highflow sensor. Then once it has been added upstream, it'll be exposed in ohmgraphite.

ok. I did opend an issue in the LibreHardwareMonitor repo. 
Lets see what happens.
Thx anyway

EnlightedBitFox, 2022-10-31:
So, now after nearly a month nothing cames around from the LibreHardwareMonitor repo.

have you any idea how i can fix it or continue?

nickbabcock, 2024-09-14:
I release OhmGraphite v0.34, which updates LibreHardwareMonitor to include high flow sensor changes: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1268

Let me know if this doesn't fix the issue.

closed 2024-09-14 by nickbabcock