nickbabcock/OhmGraphite #346: dashboard for influx not working

Kalimeiro, 2022-12-25:
Hi,

Import into Grafana for use the dashboard with influx database isn't working, it create correctly row and panel, but each panel doesn't have influx query.

I'm using latest version of Grafana/InfluxDB and OhmGraphite.

Thanks.

nickbabcock, 2022-12-26:
I just confirmed setting up a new ohmgraphite influxql dashboard with influx 2.6 that everything appears working. Please dig into the problem, and let me know the details if further help is needed.

closed 2022-12-31 by nickbabcock

Kalimeiro, 2023-02-09:
Hi,

Your confirmation is impossible, in your readme.md you specify this link to start dashboard using influxdb : https://grafana.com/grafana/dashboards/11601

And clearly, it's not working because it's for the older version of InfluxDB.

Thanks.

ausghostdog, 2023-08-20:
@Kalimeiro following this https://youtu.be/gEIgg5zHuIU?t=1282, I was having the same issue my self, you need to link the bucket in influx 2+ to a db for v1 

[image]
proof I have it working