nickbabcock/OhmGraphite #355: Sensor reading randomly disappearing

criticsnarl, 2023-02-17:
Hello, 
First, congrats to all the people involved in this product, which is a great tool to get hardware data from our different hardware stations. I am using OhmGraphite on some Windows 10 mini PCs to expose metrics to our central Prometheus server. 

I ran in some random issue of some metrics disappearing from the exposed output (prometheus output) when I restart OhmGraphite service
Taking a day long, I noticed the following:
- for some reason, one specific workstation was reset 6 times during the day (not related to OhmGraphite)
- on the first 5 reboots , OhmGraphite starts and exposes all expected metrics
- on the 6th reboot, OhmGraphite starts (all expected sensors) but when Prometheus tries to read the metrics, I get the following:
`2023-02-13 23:32:27.1698|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2023-02-13 23:33:27.1483|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/6 did not have a value
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/4 did not have a value
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/clock/1 did not have a value
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/2 did not have a value
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/1 did not have a value
2023-02-13 23:33:27.1895|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/0 did not have a value
2023-02-13 23:33:27.2011|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/5 did not have a value
2023-02-13 23:33:27.2011|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/clock/2 did not have a value`

All other metrics are correctly read and exposed.
If I restart again the OhmGraphite process, everything comes back to normal

The debug logs on service start are the following, WHEN EVERYTHING IS OK:
`2023-02-13 13:56:12.2113|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2023-02-13 13:56:12.2913|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.3.0.0, .NET 6.0.11 (6.0.11)
2023-02-13 13:56:12.3149|DEBUG|OhmGraphite.Program|Starting: parse config default
2023-02-13 13:56:12.5509|DEBUG|OhmGraphite.Program|Finished: parse config default
2023-02-13 13:56:12.5778|INFO|OhmGraphite.Program|Prometheus port: 9184
2023-02-13 13:56:12.6296|DEBUG|Topshelf.Runtime.Windows.WindowsHostEnvironment|Started by the Windows services process
2023-02-13 13:56:12.6410|DEBUG|Topshelf.Builders.RunBuilder|Running as a service, creating service host.
2023-02-13 13:56:12.6546|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|Starting as a Windows service
2023-02-13 13:56:12.6750|DEBUG|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Starting up as a windows service application
2023-02-13 13:56:12.6962|INFO|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Starting
2023-02-13 13:56:12.7061|DEBUG|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Current Directory: C:\Program Files\Ohm_Graphite
2023-02-13 13:56:12.7061|DEBUG|Topshelf.Runtime.Windows.WindowsServiceHost|[Topshelf] Arguments: 
2023-02-13 13:56:12.7142|DEBUG|OhmGraphite.PrometheusServer|Starting: starting prometheus server
2023-02-13 13:56:12.8738|DEBUG|OhmGraphite.SensorCollector|Hardware added: /motherboard
2023-02-13 13:56:12.8738|DEBUG|OhmGraphite.SensorCollector|Hardware added: /lpc/nct6776f
2023-02-13 13:56:12.8738|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/0 "Fan #1"
2023-02-13 13:56:12.8738|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/1 "Fan #2"
2023-02-13 13:56:12.8738|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6776f/control/2 "Fan #3"
2023-02-13 13:56:12.9697|DEBUG|OhmGraphite.SensorCollector|Hardware added: /intelcpu/0
2023-02-13 13:56:12.9697|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/1 "CPU Core #1 Thread #1"
2023-02-13 13:56:12.9697|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #1 Thread #2"
2023-02-13 13:56:12.9697|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #2 Thread #1"
2023-02-13 13:56:12.9697|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #2 Thread #2"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/0 "CPU Total"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Package"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #1 Distance to TjMax"
2023-02-13 13:56:12.9796|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/4 "CPU Core #2 Distance to TjMax"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/5 "Core Max"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/6 "Core Average"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/1 "CPU Core #1"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/2 "CPU Core #2"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/0 "CPU Package"
2023-02-13 13:56:12.9954|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/1 "CPU Cores"
2023-02-13 13:56:13.0115|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/0 "CPU Core"
2023-02-13 13:56:13.0115|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/1 "CPU Core #1"
2023-02-13 13:56:13.0115|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/2 "CPU Core #2"
2023-02-13 13:56:13.0115|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/0 "Bus Speed"
2023-02-13 13:56:13.0115|DEBUG|OhmGraphite.SensorCollector|Hardware added: /ram
2023-02-13 13:56:13.0115|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/0 "Memory Used"
2023-02-13 13:56:13.0255|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/1 "Memory Available"
2023-02-13 13:56:13.0255|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/0 "Memory"
[... 138 lines cut ...]
2023-02-13 17:00:40.1351|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/5 did not have a value
2023-02-13 17:00:40.1351|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/clock/2 did not have a value
2023-02-13 17:00:40.1351|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/3 did not have a value
2023-02-13 17:00:40.1351|DEBUG|OhmGraphite.PrometheusCollection|Finished: prometheus update metrics
2023-02-13 17:01:28.2933|DEBUG|OhmGraphite.PrometheusCollection|Starting: prometheus update metrics
2023-02-13 17:01:28.3159|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/6 did not have a value
2023-02-13 17:01:28.3159|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/temperature/4 did not have a value
2023-02-13 17:01:28.3159|DEBUG|OhmGraphite.SensorCollector|/intelcpu/0/clock/1 did not have a value
.....
`

OhmGraphite configuration :
`<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="9184" />
    
    
    <add key="prometheus_host" value="*" />
  </appSettings>
</configuration>
`

Many thanks

nickbabcock, 2023-02-17:
Thanks for the report. I can almost assume that the missing values are due to this [block of code in LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/50f2d6073ca7c406e845611ca4c349394aa27ecc/LibreHardwareMonitorLib/Hardware/Cpu/IntelCpu.cs#L515-L534), where it tries to read temperature and if it fails, it clears out the existing value.

Why the temperature reading fails until OhmGraphite is restarted, I am not sure. If I had to guess, it'd be due to what CPU cores and threads are considered valid as well as their order being cached upon startup. If any of this is shuffled after the initial start, I imagine that it would cause the temperature reading to fail.

Hmm keep an eye on this problem. If it's remains troublesome, I think it could be possible to workaround the issue by adding some code to re-initialize everything after persistent failures for like 10 minutes.

criticsnarl, 2023-02-20:
Hello @nickbabcock and thanks for the quick answer. If you would have to guess, why do you think the order of the cores is shuffled apparently random? I mean, would this be something under LibreHardwareMonitor control or it's how the OS presents them?

nickbabcock, 2023-02-22:
Yeah it could be in LibrehardwareMonitor, .net runtime, or the OS. I'm truly unsure. If I had to guess where to look first, I'd start with this [block of code](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/fbb7157d27e26c139a84b3720990a2acfa90cb89/LibreHardwareMonitorLib/Hardware/Cpu/CpuGroup.cs#L134-L139) and see if there is anything fishy with the code or kernel code.

nickbabcock, 2023-03-16:
There were some updates in v0.29.0 that may have addressed the issue, let me know if it is still occurring.

closed 2023-04-04 by criticsnarl