nickbabcock/OhmGraphite #372: Incorrect data reflected in influxdb

drewzeee, 2023-05-31:
I am trying to monitor the gpu temp on a windows VM, and send it to my influxdb. I have the most recent version of OhmGraphite installed on the windows VM. 

I have connected to influxdb with the following plugin:
`<?xml version="1.0" encoding="utf-8" ?>
<configuration>
	<appSettings>
    		<add key="type" value="influx2" />
		<add key="influx2_address" value="http://192.168.50.200:8086" />
		<add key="influx2_org" value="#myorg#" />
		<add key="influx2_bucket" value="ohmgraphite" />
		<add key="influx2_token" value="#mytoken#" />
		<add key="interval" value="2s" />
        </appSettings>
</configuration>`

The issue is that, while the sensor/metric types are populated correctly, the results of the queries are not. It will show the gpu core temp stuck around 45 degrees even under full load (while libre hm is showing accurate measurements via the web server). 

[image]

Given that the sensor information is populating in influxdb, I am inclined to believe that it's not an auth issue. Given that librehw monitor is reflecting accurate measurements, I am inclined to believe that it's not a hw issue. I am at a loss here.

nickbabcock, 2023-06-01:
Glad you got prometheus working, if you want to test out influxdb some more: instead of running OhmGraphite as a service, can you run it interactively (ie: `.\OhmGraphite.exe run`) and see if the issue persists? There's been a few issues that only seem to show when running as a service.

nickbabcock, 2023-11-13:
Closing, but happy to continue troubleshooting as needed.

closed 2023-11-13 by nickbabcock