nickbabcock/OhmGraphite #390: Example Dashboard from README.md

TechnoSwiss, 2023-07-12:
The example dashboards are great, but the image you have in the README.md file looks much cleaner. Is that dashboard available somewhere?

Thanks.

nickbabcock, 2023-07-12:
The dashboard in the readme is the prometheus example dashboard. You're right that the other dashboards are a bit lackluster in comparison. It's just a labor intensive task to port the dashboard to other data sources, especially as the prometheus dashboard was in flux for so long. Related issue and comment: #246 / https://github.com/nickbabcock/OhmGraphite/issues/368#issuecomment-1534000530

I will get around to it eventually, but if someone wants to champion the effort of porting the prometheus dashboard, I will be extremely grateful.

TechnoSwiss, 2023-07-12:
Shoot, I looked through all the open issues to make sure this hadn't been asked already and missed that one somehow. I may take a crack and updating that for InfluxDB.

closed 2023-07-12 by TechnoSwiss